NVIDIA-Omniverse/PhysX #25: Compile error with Ubuntu 20.04 and g++11, but works with clang 14

Ravbug, 2022-11-11:
### Library and Version

PhysX v5.1.0 (release/104.0)

### Operating System

- OS: Ubuntu 20.04
- Compiler: g++ 11, clang 14
- C++ version: C++17

### Steps to Trigger Behavior

 1. Compile `PhysX` target using clang 14 - compilation succeeds
 2. Compile `PhysX` target using g++ 11 - error when building `PxAutoGeneratedMetaDataObjects.cpp` due to macro expansion of `DEFINE_PROPERTY_TO_VALUE_STRUCT_MAP` in `PxMetaDataObjects.h`:
```cpp
physx/physx/source/physxmetadata/core/src/PxAutoGeneratedMetaDataObjects.cpp
In file included from physx/physx/source/physxmetadata/core/src/PxAutoGeneratedMetaDataObjects.cpp:40:
physx/physx/source/physxmetadata/core/include/PxMetaDataObjects.h:498:87: error: expected unqualified-id before ‘)’ token
  498 |                 PxPropertyToValueStructMemberMap< PxPropertyInfoName::type##_##prop >() : Offset( PX_OFFSET_OF_RT( valueStruct, prop ) ) {}             \
      |                                                                                       ^
physx/physx/source/physxmetadata/core/include/PxAutoGeneratedMetaDataObjects.h:60:9: note: in expansion of macro ‘DEFINE_PROPERTY_TO_VALUE_STRUCT_MAP’
   60 |         DEFINE_PROPERTY_TO_VALUE_STRUCT_MAP( PxPhysics, TolerancesScale, PxPhysicsGeneratedValues)
```

### Code snippets
`cmake --build . --target PhysX`

### Expected Behavior

g++11 should successfully compile the `PhysX` target

### Actual Behavior

g++11 is not able to compile the `PhysX` target

Ravbug, 2022-11-18:
Also fails in the same way with g++12 on Ubuntu 22.04.

jcarius-nv, 2022-11-22:
Thanks for reporting. While I don't see your exact error, compilation with gcc11 indeed fails. We've narrowed down the issue and it will be fixed in the next version.

As a workaround for now, you can try adjusting the build flags for gcc, in particular remove `-Werror` and add `-fpermissive`. The easiest place to change this is in `source/compiler/cmake/linux/CMakeLists.txt` at the line that begins with `SET(GCC_WARNINGS ...`

Ravbug, 2022-12-20:
I tried `-fpermissive` and updated to the latest PhysX release but I am still getting the error. Here is the full command that CMake is running to compile `PxMetaDataObjects.h`:
> ``` /usr/bin/g++-11 -DNDEBUG -DPX_PHYSX_GPU_SHARED_LIB_NAME=libPhysXGpu_64.so -DPX_PHYSX_STATIC_LIB -DPX_PUBLIC_RELEASE=1 -DPX_SUPPORT_OMNI_PVD=0 -DPX_SUPPORT_PVD=0 -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/common/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/common/src -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/physx/src -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/physx/src/device -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/physxgpu/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/contact -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/common -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/convex -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/distance -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/sweep -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/gjk -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/intersection -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/mesh -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/hf -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/pcm -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/geomutils/src/ccd -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/lowlevel/api/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/lowlevel/software/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/lowlevel/common/include/pipeline -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/lowlevel/common/include/utils -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/lowlevelaabb/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/lowleveldynamics/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/simulationcontroller/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/simulationcontroller/src -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/scenequery/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/physxmetadata/core/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/immediatemode/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/pvd/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/omnipvd -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/pvdruntime/include -I/home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/foundation/include -std=c++11 -fno-rtti -fno-exceptions -ffunction-sections -fdata-sections -fno-strict-aliasing  -Wall -Werror		-Wno-array-bounds		-Wno-class-memaccess		-Wno-invalid-offsetof		-Wno-mismatched-new-delete		-Wno-nonnull-compare		-Wno-pragmas		-Wno-uninitialized		-Wno-unused-function	 -O3 -fPIC -w -std=gnu++20 -MD -MT deps/physx/physx/compiler/public/sdk_source_bin/CMakeFiles/PhysX.dir/__/__/physxmetadata/core/src/PxMetaDataObjects.cpp.o -MF deps/physx/physx/compiler/public/sdk_source_bin/CMakeFiles/PhysX.dir/__/__/physxmetadata/core/src/PxMetaDataObjects.cpp.o.d -o deps/physx/physx/compiler/public/sdk_source_bin/CMakeFiles/PhysX.dir/__/__/physxmetadata/core/src/PxMetaDataObjects.cpp.o -c /home/runner/work/RavEngine/RavEngine/deps/physx/physx/source/physxmetadata/core/src/PxMetaDataObjects.cpp ```

jcarius-nv, 2022-12-20:
This is surprising because we're testing the release in a clean containerized Ubuntu environment the respective default clang and gcc versions, i.e., Ubuntu 20.04 LTS with gcc 9 or clang 10 and Ubuntu 22.04 LTS with gcc 11 or clang 14, and have not seen any problems.

Are you using a customized setup to build PhysX?

Your original post indicates that the compile error happens when compiling `PxAutoGeneratedMetaDataObjects.cpp`. Your most recent post shows the build flags for `PxMetaDataObjects.cpp`. Just copying them for readability, the flags are
```
-std=c++11 -fno-rtti -fno-exceptions -ffunction-sections -fdata-sections -fno-strict-aliasing 
-Wall -Werror		-Wno-array-bounds		-Wno-class-memaccess		-Wno-invalid-offsetof		
-Wno-mismatched-new-delete		-Wno-nonnull-compare		-Wno-pragmas		-Wno-uninitialized		
-Wno-unused-function	 -O3 -fPIC -w -std=gnu++20 -MD -MT
```
and do not include `-fpermissive` as far as I can tell. Can you double-check that the flags for the actual file that produces errors are set correctly? I know the error happens in the header file, but that header is included from multiple source files, so it's important that you're actually working with the right cpp file.

Once you figured out which file exactly it is and ruled out other problems, you can apply a patch similar to the lines in [source/compiler/cmake/linux/PhysX.cmake#L122](https://github.com/NVIDIA-Omniverse/PhysX/blob/release/104.1/physx/source/compiler/cmake/linux/PhysX.cmake#L122), i.e., something like
```Cmake
IF ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "GNU")
  SET_SOURCE_FILES_PROPERTIES(${PX_SOURCE_DIR}/omnipvd/OmniPvdPxSampler.cpp PROPERTIES COMPILE_FLAGS "-fpermissive -Wno-error")
ENDIF()
```

jcarius-nv, 2023-01-05:
Great, thanks for letting us know!

closed 2023-01-05 by jcarius-nv