NVIDIA-Omniverse/PhysX #62: Missing triangle using renderSoftBodyGeometry

rwang15, 2022-12-22:
`SnippetRender::renderSoftBodyGeometry` only render three faces for each tetrahedron as `tetFaces` uses incorrect vertice indexs. 

https://github.com/NVIDIA-Omniverse/PhysX/blob/5044420f72788fe48ab92ffceb66fa0be18cb851/physx/snippets/snippetrender/SnippetRender.cpp#L124

should be 
```
	const int tetFaces[4][3] = { {0,2,1}, {0,1,3}, {0,3,2}, {1,2,3} };
```

preist-nvidia, 2023-01-03:
Thank you @rwang15, we will have a look.

nvtw, 2023-01-09:
Thanks for noticing. I just submitted a fix to our internal development version. It will probably take a while until it shows up on github but I hope you can fix it locally in your version in the meantime.

closed 2023-10-09 by rwang15