NVIDIA-Omniverse/PhysX #65: Unable to create D6 joint from separate library when statically linking PhysX

Zeblote, 2022-12-24:
I just stumbled over this issue while testing my Unreal Engine integration. It seems that if you use the static libraries for PhysX, and attempt to use PxD6JointCreate in a different dll than the one that initialized PhysX, it will crash because FdFoundation::mInstance is null in that dll when trying to allocate the joint. The PxPhysics passed into PxD6JointCreate could be used to find the correct foundation, but it seems to be ignored.

In my case the fix is simple - just use the dynamic library instead for modular builds - but I saw other fixes in the changelog for this exact configuration with multiple dlls using static PhysX so I thought I should bring it up here.

preist-nvidia, 2023-01-03:
Thank you @Zeblote, we'll have a look.

Freznosis, 2023-03-19:
So is using static libraries not possible for this use case? Running into this issue as well.

PierreTerdiman, 2023-03-20:
Mixing PhysX static libraries and DLLs is not recommended, as it opens the door for these issues. One thing you could try is use PxSetFoundationInstance to initialize the faulty foundation pointer with the correct one.

One thing I wonder is why the SDK did not assert for you in PxInitExtensions? If I understand correctly, I think this test there should have failed. But maybe you didn't run it in Debug.

bool PxInitExtensions(PxPhysics& physics, PxPvd* pvd)
{
	PX_ASSERT(&physics.getFoundation() == &PxGetFoundation());

closed 2023-10-27 by preist-nvidia