NVIDIA-Omniverse/PhysX #72: macOS build of blast checks for wrong uname output

jordanhubbard, 2023-01-12:
% cd blast
% ./build.sh 
[ERROR][omni.repo.man.guidelines] Unsupported host platform: darwin-x86_64, only ['windows-x86_64', 'linux-x86_64', 'linux-aarch64', 'macos-x86_64', 'macos-aarch64'] are supported.

% uname -a
Darwin jordanh-mlt 22.1.0 Darwin Kernel Version 22.1.0: Sun Oct  9 20:15:09 PDT 2022; root:xnu-8792.41.9~2/RELEASE_ARM64_T6000 arm64

This is an Apple Silicon Mac so both the output (claiming a darwin-x86_64 host platform) and the uname check is erroneous.  macos has always identified itself as "Darwin" due to its roots, never "macos".

nvidia-bgaldrikian, 2023-01-12:
Hi jordanhubbard,

Unfortunately we don't currently support a MacOS build of blast.  I'm pretty sure that platform name was carried through some ancient build tools accidently.  We'll at least scrub it from the list of supported platforms.  I'm sorry for the confusion.

Bryan Galdrikian

jordanhubbard, 2023-01-12:
Is blast in the critical path for Omniverse?  I'm not sure how much of the PhysX project is currently consumed by OV, but we do need macos / iOS support for the pieces that are.

closed 2023-10-27 by preist-nvidia