NVIDIA-Omniverse/PhysX #73: flow build.sh is hard-coded with behaviors that will never work on macos

jordanhubbard, 2023-01-12:
### Library and Version
PhysX ToT (main branch)

### Operating System
macOS 22.1.0 ARM
### Steps to Trigger Behavior

 1. cd flow
 2. ./build.sh

### Code Snippet to Reproduce Behavior

```
#! /bin/bash

./generate_projects.sh

cd _compiler/gmake2

make config=debug_x86_64
make config=release_x86_64

cd ../..

```

### Expected Behavior

A build script that does not hard-code x86_64 or assume that gmake2 is already built (the existing build.sh appears to work for only one Linux target).
### Actual Behavior

./generate_projects.sh: line 3: ./external/premake/linux64/premake5: cannot execute binary file
./build.sh: line 5: cd: _compiler/gmake2: No such file or directory
make: *** No targets specified and no makefile found.  Stop.
make: *** No targets specified and no makefile found.  Stop.

erwincoumans, 2023-02-03:
The build system was mostly used and tested on Windows and Linux (Ubunbu) and the MacOS build needs more attention. We will follow up once there is news.

preist-nvidia, 2023-10-27:
This will be addressed when we update platform requirements.

closed 2023-10-27 by preist-nvidia