NVIDIA-Omniverse/PhysX #75: Assert removing aggregate with static filter hint set that overlaps other static aggregates and shapes

Zeblote, 2023-01-15:
Hi, I've just fixed this bug so I thought you might want to take the fix. Persistent pair is only created conditionally in `AABBManager::processBPCreatedPair` based on their filter hints, however, `AABBManager::processBPDeletedPair`
 expects a pair to always exist in the map, and thus causes the assert.

There is probably a better way that skips searching the map entirely based on the filter hints. So I won't create a PR with my bodge.
https://github.com/brickadia/PhysX/commit/204b5130eb8c2d599fe0867a445f4550d84d66fc

PierreTerdiman, 2023-01-20:
Thank you, I see what happened and it looks like a legitimate bug. I will try to create a repro and investigate what the best fix would be.

PierreTerdiman, 2023-01-20:
Yes this works:

```
	const AggPairMap::Entry* e = pairMap->find(AggPair(volA, volB));
	if(e)
	{
		PersistentPairs* p = e->second;
		p->outputDeletedOverlaps(mDestroyedOverlaps, mVolumeData.begin());
		p->mShouldBeDeleted = true;
	}

```

Thank you.

preist-nvidia, 2023-10-27:
Done, fix in 5.1.3.

closed 2023-10-27 by preist-nvidia