NVIDIA-Omniverse/PhysX #83: Assert in suspension on flipped vehicle.

Tooflevor, 2023-01-31:
It seems like `suspDirVelWheel` should be used in line 373.
https://github.com/NVIDIA-Omniverse/PhysX/blob/4d33492253ab64c2246d80257ac4ce0f37b40b22/physx/source/physxvehicle2/src/suspension/VhSuspensionFunctions.cpp#L365-L376
Otherwise, the assert in line 376 is not guaranteed for cases where `-gravitySuspDir * dt * dt > previousJounce + jounceSpeed * dt`. It is possible case when vehicle is flipped.

### Library and Version
PhysX v5.1.2

### Operating System
Window 10

msauter-nvidia, 2023-02-01:
Thanks for pointing this out. It's a good point. Will create ticket to look at it. However, I would not use suspDirVelWheel inside the if-statement since that is more of a heuristic to not be too aggressive with clamping. But the check should probably change to not accidentally end up with a negative clamped jounce.

msauter-nvidia, 2023-02-22:
Will get fixed in PhysX 5.2.0

closed 2023-02-22 by msauter-nvidia