NVIDIA-Omniverse/PhysX #85: Undefined Symbols during runtime

farbod1277, 2023-02-07:
Hi there,

I'm trying to use the PhysX API within a ROS2 node. I've cloned this repository and followed the build instructions. I'm using Ubuntu 20.04 LTS and have built the "checked" version of PhysX. All the snippets run without any issues.

The instructions for building applications with PhysX refer to adding the /include and /bin folder to the project's makefile but since ROS2 auto generates the makefiles I have to do it at the package's CMakeLists.txt level. Hence I've added the following to my CMakeLists.txt:

```
include_directories($ENV{PHYSX_INCLUDE})
include_directories($ENV{PHYSX_BIN})

target_link_libraries(vehicle_simulator_node
    ${PROJECT_NAME}
    $ENV{PHYSX_LIB}/libPhysXCharacterKinematic_static_64.a
    $ENV{PHYSX_LIB}/libPhysXCommon_static_64.a
    $ENV{PHYSX_LIB}/libPhysXCooking_static_64.a
    $ENV{PHYSX_LIB}/libPhysXExtensions_static_64.a
    $ENV{PHYSX_LIB}/libPhysXFoundation_static_64.a
    $ENV{PHYSX_LIB}/libPhysXGpu_64.so
    $ENV{PHYSX_LIB}/libPhysXPvdSDK_static_64.a
    $ENV{PHYSX_LIB}/libPhysX_static_64.a
    $ENV{PHYSX_LIB}/libPhysXVehicle2_static_64.a
    $ENV{PHYSX_LIB}/libPhysXVehicle_static_64.a
    $ENV{PHYSX_LIB}/libPVDRuntime_64.so
)
```
Here is the output of env:
```
PHYSX_ROOT=/home/ubuntu/PhysX
PHYSX_BIN=/home/ubuntu/PhysX/physx/bin
PHYSX_LIB=/home/ubuntuPhysX/physx/bin/linux.clang/checked
PHYSX_INCLUDE=/home/ubuntu/PhysX/physx/include
```
I've also added the following lines to my application's header file where I'd be initialising PhysX:
```
#define NDEBUG
#include "PxPhysicsAPI.h"
```
Within this file I have the following lines for initialising PhysX:
```
static PxDefaultErrorCallback gDefaultErrorCallback;
static PxDefaultAllocator gDefaultAllocatorCallback;
PxFoundation *mFoundation;
PxPhysics *mPhysics;
```
And when I run my ROS2 node I get the following error:
```
undefined symbol: _ZN319VehicleSimulator16VehicleSimulator21gDefaultErrorCallbackE, at /tmp/binarydeb/ros-foxy-rcutils-1.1.4/src/shared_library.c:84
```
This suggests that there was an issue with properly linking the PhysX libraries, but I'm not sure what I've done wrong. Any help would be much appreciated. Please let me know if you need any more details. Thanks.

jcarius-nv, 2023-02-07:
I've seen something like this before, we might have possibly introduced a circular dependency in the PhysX libraries so they need to be linked multiple times. As a workaround, you can let the linker take multiple iterations over all the libraries until all the symbols are resolved. Something like this should work

```
target_link_libraries(Snippet
    -Wl,--start-group # put physx libraries in linker group due to circular dependencies
        PhysX_static_64
        PhysXCommon_static_64
        PhysXExtensions_static_64
        PhysXFoundation_static_64
        PhysXPvdSDK_static_64
    -Wl,--end-group

    pthread
    dl
)
```

farbod1277, 2023-02-07:
@jcarius-nv that worked! Amazing, thank you

farbod1277, 2023-02-07:
@jcarius-nv ok so it now works but only if I don't define the errorCallback and allocatorCallback as static. The same error occurs still when they're defined as static. Thanks

ayoub-belarbi, 2023-06-21:
Hello @farbod1277, 
Two things that you may try to solve this if you haven't already:
1. Add PUBLIC to target_link_libraries: By doing that, You ensure that the specified PhysX libraries are linked not only to vehicle_simulator_node, but also to any targets that depend on vehicle_simulator_node. That way, the linked libraries and their dependencies are available throughout the dependency tree, which can help in resolving linking issues and ensuring correct library usage.
2. Adjust Linking Order: The order in which libraries are linked can affect symbol resolution. Try rearranging the order of the libraries in the target_link_libraries command. Place the libraries with dependencies (e.g., libPhysX_static_64.a) before the dependent libraries (e.g., libPhysXCharacterKinematic_static_64.a). This is why @jcarius-nv suggested adding the linker group.

Anyway since it has been a while since this has been opened. I will close it now. If you have any input from your side, feel free to open it again.
Thanks

closed 2023-06-21 by ayoub-belarbi