NVIDIA-Omniverse/PhysX #86: Negative TOI calculation problem for Convex- TriangleMesh CCD

pkusilence, 2023-02-09:
[image]

https://github.com/NVIDIA-Omniverse/PhysX/blob/release/104.1/physx/source/geomutils/src/ccd/GuCCDSweepConvexMesh.cpp

In GuCCDSweepConvexMesh.cpp, Line 555 to Line 584, it is calculating the penetration if TOI < 0 after SweepShapeTriangle. However, the distanceSq makes me confused. Since sphereCenterInTransform1 is always identity （trimesh must be static). So the distance is from triangle mesh vertex space origin to the hit triangle, but it is arbitrary，since triangle mesh vertex space origin can be anywhere. So the TOI might be wrong in this case.

preist-nvidia, 2023-12-11:
Hi @pkusilence - sorry, we dropped the ball here, my bad.

Did you figure out more in the mean time or would it be helpful if we had a closer look still?

Internal tracking: PX-4673

preist-nvidia, 2024-01-08:
Closing for now until further comments are made.

closed 2024-01-08 by preist-nvidia

pkusilence, 2024-01-24:
> Hi @pkusilence - sorry, we dropped the ball here, my bad.
> 
> Did you figure out more in the mean time or would it be helpful if we had a closer look still?
> 
> Internal tracking: PX-4673

I choosed to skip all the things PhysX done for negative TOI, the simulation looks fine now. Thanks for your reply!