NVIDIA-Omniverse/PhysX #88: Werror flag is making impossible to compile in Linux (Manjaro)

baziorek, 2023-02-11:
### Library and Version
4d33492253ab64c2246d80257ac4ce0f37b40b22

### Operating System
Manjaro 22.0.2
gcc version 12.2.1 20230111 (GCC) 

### Steps to Trigger Behavior

 1. `git clone https://github.com/NVIDIA-Omniverse/PhysX.git`
 2. `cd PhysX/physx`
 3. `generate_projects.sh`
 4. `cd compiler/linux-debug/ && make`

Here is error message:
```
/physx/source/lowlevel/software/src/PxsCCD.cpp:1767:8: error: variable 'totalActivePairs' set but not used [-Werror,-Wunused-but-set-variable]
        PxU32 totalActivePairs = 0;
```

### Code Snippet to Reproduce Behavior

```
<your-snippet>
```

### Expected Behavior
Everything compiles

### Actual Behavior
Warning became compile error - it does not compile

### Fast fix
I fixed its by removing flag `-Werror` from file `./PhysX/physx/source/compiler/cmake/linux/CMakeLists.txt`

Zeblote, 2023-02-11:
A better fix would be to remove the unused variable or ignore that specific warning: `-Wno-unused-but-set-variable`

LordOfDragons, 2023-02-11:
There are many such places with unused variables. Trying to fix them all would be a chore. Needs a fix upstream or allowing to disable this warning for release builds.

Opting for proper fix though. I always build with Wall Werror and just can't understand how such things can end up in a release.

Zeblote, 2023-02-11:
There were recent changes to the behavior of `unused-but-set-variable`, to warn in additional cases, so maybe they only tested PhysX with an older compiler version.

jcarius-nv, 2023-02-13:
Please see https://github.com/NVIDIA-Omniverse/PhysX/blob/release/104.1/physx/documentation/platformreadme/linux/README_LINUX.md for supported platforms and compiler versions, in your case that would be
```
    Ubuntu 20.04 LTS with gcc 9 or clang 10
    Ubuntu 22.04 LTS with gcc 11 or clang 14
```
For anything newer I'm afraid we can't guarantee that it just works out of the box. As @Zeblote mentions, you're free to disable the warning.

LordOfDragons, 2023-02-13:
Using "gcc 11.3.1" here and it causes many warnings failing to build. According to your statement gcc 11 is supported and should work so in my opinion this is a clear bug then... unless I'm misunderstanding something here.

jcarius-nv, 2023-02-13:
@LordOfDragons It should build, yes. Can you share some more details about your setup, and the build log that shows the compiler invocation and the error?

LordOfDragons, 2023-02-13:
Which file are you looking for?

jcarius-nv, 2023-02-13:
Similar to the information that was provided in https://github.com/NVIDIA-Omniverse/PhysX/issues/25, i.e., the exact error message and the gcc command that `make` is calling so we can see all the flags passed to it.
I think you should be able to read all that in the console log if you run `make VERBOSE=1`

LordOfDragons, 2023-02-13:
I can't find any file mentioned there to post so I piped the console output into a file: https://pastebin.com/1t4CFGJA . I do though not see the compile parameters in there so is there some kind of log file written which contains this information? As you can see I've modified the build files to not raise an error for each warning which is why you can see all of the problems.

jcarius-nv, 2023-02-13:
interesting, it appears that the same minor version of gcc behaves differently on different linux flavors (?!).
I just checked again and can confirm that we're testing on `gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0`

LordOfDragons, 2023-02-14:
Well... there is a small difference (11.3.1 -> 11.3.0) but I don't think that's to blame.