NVIDIA-Omniverse/PhysX #90: Getting NaN link velocities with a simple PxArticulationReducedCoordinate

cjds, 2023-02-15:
### Library and Version
PhysX 5.1.0

### Operating System
Linux

### Code Snippet to Reproduce Behavior

```
physx::PxArticulationReducedCoordinate* create_single_joint_articulation(physx::PxPhysics& physics)
{
    physx::PxArticulationReducedCoordinate* articulation = physics.createArticulationReducedCoordinate();
    articulation->setArticulationFlags(physx::PxArticulationFlag::eFIX_BASE);
    material = physics.createMaterial(0.5f, 0.8f, 0.3f);

    // essentially a big rotating cube (1m)
    const physx::PxBoxGeometry link_geometry = physx::PxBoxGeometry(1.0);
    // add a second link still at identity
    auto root_link = articulation->createLink(nullptr, physx::PxTransform());
    physx::PxRigidActorExt::createExclusiveShape(*root_link, link_geometry, *material);

    // add a second link still at identity
    auto link = articulation->createLink(root_link, physx::PxTransform(1,0,0));
    physx::PxRigidActorExt::createExclusiveShape(*link, link_geometry, *material);
    physx::PxTransform on_x(1,0,0);
    // No need to fiddle with the joint as we're not setting limits
    auto joint = static_cast<physx::PxArticulationJointReducedCoordinate*>(link->getInboundJoint());
    joint->setParentPose(on_x);
    joint->setJointType(physx::PxArticulationJointType::eREVOLUTE);
    joint->setMotion(physx::PxArticulationAxis::eTWIST, physx::PxArticulationMotion::eFREE);
    return articulation;
}

void printVel(const physx::PxSpatialVelocity& t)
{
  std::cout << "v: [" << t.linear.x << "," << t.linear.y << "," << t.linear.z << "]";
  std::cout << "w: [" << t.angular.x << "," << t.angular.y << "," << t.angular.z << "]" << std::endl;
}

int main()
{
    physx::PxDefaultAllocator allocator;
    physx::PxDefaultErrorCallback error_callback;
    physx::PxFoundation* foundation = PxCreateFoundation(PX_PHYSICS_VERSION, allocator, error_callback);
    physx::PxPhysics* physics = PxCreatePhysics(PX_PHYSICS_VERSION, *foundation, physx::PxTolerancesScale(), true, nullptr);
    physx::PxSceneDesc scene_desc(physics->getTolerancesScale());
    physx::PxScene* scene = physics->createScene(scene_desc);

    auto art = create_single_joint_articulation(*physics);

    scene->addArticulation(*robot);
    art ->zeroCache(*cache);
    cache->jointVelocity[0] = 1;
    art ->applyCache(*cache, physx::PxArticulationCacheFlag::eALL);
    art ->copyInternalStateToCache(*cache, physx::PxArticulationCacheFlag::eALL);
    printVel(cache->linkVelocity[1]);
}
```

### Expected Behavior

I would expect the velocity of the second link to be some non-zero value if the first joint is moving at 1m/s

### Actual Behavior
The output of the above program is

```
v: [nan,-nan,-nan]w: [-nan,-nan,-nan]
```

I don't understand why the link velocities in this case will ever be NaN

preist-nvidia, 2023-02-16:
Hi, I can't see an obvious mistake from the snippet, but it also looks incomplete. Could you please adapt the articulation snippet with your repro and provide it? Then I can have a look.

https://github.com/NVIDIA-Omniverse/PhysX/blob/release/104.1/physx/snippets/snippetarticulationrc/SnippetArticulation.cpp

cjds, 2023-02-16:
Yes. I'll give that a try and get back to you by Monday

preist-nvidia, 2023-10-27:
Closing as did not get repro.

closed 2023-10-27 by preist-nvidia