NVIDIA-Omniverse/PhysX #102: CUDA Rigidbodies extremely unstable when initializing with setGlobalPose instead of setLocalPose

tysoni, 2023-02-26:
If you initialize CUDA rigidbodies in certain configurations with setGlobalPose instead of setLocalPose, they will behave erratically. The erratic behavior does not happen in CPU mode.

This bug report is an extension of a discussion I started last week. I am logging it as a bug because I have created a fully reproducible snippet showcasing the erratic behavior.

In my snippet I create a scene, simulate for a couple of time steps, then cleanup the scene and repeat that process many times - only displaying the final result after all timesteps are simulated.

When I initialize the rigidbodies by establishing their global pose at the scene origin, and moving their shape local pose into the desired place in the world, the simulation is fully deterministic.

When I do the opposite - establish their shape local pose at the rigidbody origin and move their rigidbody global pose into the desired place in the world, the simulation behaves erratically.

The video below demonstrates the result: 

https://user-images.githubusercontent.com/2181360/221393423-e3f3ca3f-87bd-45d9-ac12-013918d34f37.mp4

And here is the snippet. Replace the existing files in the PhysX "SnippetHelloGRB" project with these two, and uncomment lines 248/249 to see the erratic behavior.

[unstableCUDA.zip](https://github.com/NVIDIA-Omniverse/PhysX/files/10833012/unstableCUDA.zip)

[image]

Note: The rigidbodies in this snippet are initialized in a manner that causes them all to slightly overlap on their start frame. This is by design, as the erratic behavior occurs most often when starting penetrations exist, in my tests.

PierreTerdiman, 2023-02-27:
Notes so far:

If you run this in Debug you quickly hit asserts triggered by NaNs in the system. It is likely that they are the source of non-determinism. Now the question is why/where they're coming from - I haven't figured it out yet.

The issue seems to vanish if you set the mass/inertia properly, as I think you tried to do in commented out lines.

It only seems to affect the PGS solver running on the GPU.

Note that local & global poses are usually not interchangeable. The local poses are mainly used to create 'compounds', so for simple bodies with a single shape like in your test, the proper way would be an identity local pose and a non-identity global pose, i.e. the version that doesn't work here.

jcarius-nv, 2023-03-29:
Hi @tysoni 
I managed to trace down the issue to a buffer overflow problem in GPU box-box contact generation. With the fix applied your snippet remains deterministic. The problem should, therefore, be solved in the next release.

Thanks again for reporting and providing a full reproduction!

tysoni, 2023-03-29:
That's great news, @jcarius-nv, although I also get the issue with convex hulls (not sphere hulls from what I can see). Would the buffer overrun fix for box-box contacts affect convex hulls as well? Or would that be a separate overrun?

jcarius-nv, 2023-03-29:
Do you have, by any chance, another repro available for that case? I could then check the exact code path taken.

tysoni, 2023-04-03:
@jcarius-nv Hmm, so I thought I had one, but looking closer at the setup (it's a bit complex) the issue may be related to a box-collider fallback I use for degenerate input geometry. In simpler setups I am not able to cause indeterminism when I force convex colliders for all meshes.

So, I'm going to mark this issue as closed and will re-open a separate issue if I run into further indeterminism after the next build is released. Thanks!

closed 2023-04-03 by tysoni