NVIDIA-Omniverse/PhysX #113: postFilter() not work properly in PxSceneQueryExt::overlapMultiple()

SHiziw, 2023-03-10:
### Library and Version
PhysX v5.1.2

### Operating System
Example: Windows 10 20H2

### Steps to Trigger Behavior

 1. Derive `PxQueryFilterCallback::postFilter(const PxFilterData& filterData, const PxQueryHit& hit, const PxShape* shape, const PxRigidActor* actor)`, return `PxQueryHitType::eNONE` for unexpected actors.
 2. Raise flag: `xQueryFlag::ePOSTFILTER`
 3. Call `PxSceneQueryExt::overlapMultiple(*m_pScene, geometry, pose, hitBuffer, bufferSize, filterData, filterCall);`

### Code Snippet to Reproduce Behavior

```

PxQueryHitType::Enum SphericalConePostFilter::postFilter(const PxFilterData& filterData, const PxQueryHit& hit,
    const PxShape* shape, const PxRigidActor* actor)
{
    PX_UNUSED(filterData);
    PX_UNUSED(hit);
    const PxGeometry& testGeom = shape->getGeometry();
    PxTransform actorGlbPse = actor->getGlobalPose();
    PxTransform geometryGlobalTm = actorGlbPse.transform(shape->getLocalPose());

    DefSphericalCone* cone = new DefSphericalCone(m_openAngle, m_radius, m_margin);

    switch (int(testGeom.getType()))
    {
    case PxGeometryType::eSPHERE:
    case PxGeometryType::eCAPSULE:
    case PxGeometryType::eBOX:
    case PxGeometryType::eCONVEXMESH:
    {
        PxGjkQueryExt::ConvexGeomSupport geomSupport(testGeom);

        if (PxGjkQuery::overlap((*cone), geomSupport, m_globalPose, geometryGlobalTm))
        {
            delete cone;
            return PxQueryHitType::eTOUCH;
        }
        break;
    }
    default:
        NX_ASSERT(false);
        break;
    }

    delete cone;
    return PxQueryHitType::eNONE;
}

...

 PxQueryFilterData filterData;
filterData.data = filter;
filterData.flags = (PxQueryFlag::PxQueryFlag::eSTATIC | PxQueryFlag::eDYNAMIC | PxQueryFlag::ePOSTFILTER | PxQueryFlag::eNO_BLOCK);

...

m_hitCount = PxSceneQueryExt::overlapMultiple(*m_pScene, geometry, pose, hitBuffer, bufferSize, filterData, filterCall);

```

### Expected Behavior

`SphericalConePostFilter()` is the derived class for `PxQueryFilterCallback()` so since I return `PxQueryHitType::eNONE` in `postFilter()`, the overlap query `overlapMultiple()` should ignore it and return 0.

### Actual Behavior

`overlapMultiple()` still return 1, even the `postFilter()` return `PxQueryHitType::eNONE`.

PierreTerdiman, 2023-03-13:
Interesting. I think this is a bug indeed, and I have to congratulate you for finding one that has been here for more than a decade without anybody noticing.

I *think* the fix is to modify the line here (https://github.com/NVIDIA-Omniverse/PhysX/blob/release/104.2/physx/source/scenequery/src/SqQuery.cpp#L593) like this:

```
			if(mNoBlock && hitType==PxQueryHitType::eBLOCK)
				hitType = PxQueryHitType::eTOUCH;

```

preist-nvidia, 2023-10-27:
Fixed in 5.2.1

closed 2023-10-27 by preist-nvidia