NVIDIA-Omniverse/PhysX #126: gjkPenetration() crash

tysoni, 2023-03-24:
I've run into a situation where gjkPenetration() (the function at line 98 of GuGJKPenetration.h) fails to calculate a proper penetration info for colliding convex hulls, which leads to an assert being thrown in debug mode (and later stack corruption due to out-of-bounds array access).

Unfortunately I don't really have a way of providing a snippet for the crashing setup, but the ultimately cause of the assert and fix are easy to understand:

On the 4th iteration of the loop at line 204, 'closest' is returning from 'GJKCPairDoSimplex' as a zero vector (0,0,0,0). That leads to 'v' turning into a NaN vector (-nan, -nan, -nan, -nan) due to the division by zero in the call to V3ScaleInv on line 265. The NaN vector prevents the loop from breaking, which means in future iterations where 'size' is >= 5 the assert on line 259 is triggered. And of course accessing the prior arrays with an index >= 5 causes some out-of-bounds stack corruption, leading to a crash.

I wasn't able to solve the zero vector mystery (improperly cooked convex hull? The cooking function itself returned 'success' on the geometry, but who knows)....but the fix is easy: by checking if 'closest' is a zero vector before the call to V3ScaleInv and returning 'GJK_DEGENERATE' if it is, the crash is averted....and in the simulation itself the failed penetration calculation doesn't seem to really have a visible effect on the result. 

Either way it would be nice to return cleanly from that function in the case of bad inputs, rather than crashing.

Zeblote, 2023-04-01:
I tried to add this code you suggested, and it completely broke PxGjkQuery::proximityInfo in my tests. Suddenly no longer returning overlap for shapes that are overlapping.

tysoni, 2023-04-01:
Interesting, I would have thought that NaN output from the ScaleInv function would always cause issues but I guess you found a case where it doesn't (perhaps size < 4 when the vector is zero and in the next iteration of the loop the vector is not zero).

A more foolproof condition would be to check if size > 4 right after the assert and return GJK_DEGENERATE if so....because if it is > 4 then the next iteration of the loop is guaranteed to corrupt the stack due to array out of bounds access.

vreutskyy, 2023-04-04:
Hi @tysoni. Thanks for the report.
It's hard to say without debugging your exact situation, but looking into the code it seems that a zero vector (0,0,0,0) it valid output for 'GJKCPairDoSimplex', and it means that the shapes are penetrating. In this case it's perhaps just some zero division check that's missing in line 265.
Something like
`			v = V3Sel(FIsGrtr(dist, eps), V3ScaleInv(closest, dist), closest);
`
Could you try it?
Thanks.

tysoni, 2023-04-04:
Hi @vreutskyy,

That may alleviate the division by zero issue but in my case the zero vector is being returned on the fourth iteration of the loop, and it's not satisfying the loop exit condition, which means the stack still gets corrupted in the next iteration when the out-of-bounds array access occurs. 

I apologize for not being able to provide a snippet here...the crash is occurring in a complex setup. Either way, the fact that the loop can still continue iterating even when the "size" index value is invalid is a problem.

vreutskyy, 2023-04-04:
@tysoni 
As I see eps comes from here
`		const FloatV minMargin = FMin(a.ConvexA::getMinMargin(), b.ConvexB::getMinMargin());
		const FloatV eps = FMul(minMargin, FLoad(0.1f));
`
Do you have negative margin on one of you shape? Margin should never be negative if I'm not wrong.

tysoni, 2023-04-04:
Ah...interesting. The margin calculation is indeed negative, but it's not something I ever assign. However, it's dependent on the convex scale value, which is negative in my case due to a negative scale value on the geometry in my scene that I'm assigning to the convex mesh geometry. So I think that may actually be the root cause of all these issues!

What's interesting is that the negative scale doesn't seem to have an impact on the sim outside of these edge cases preventing the loop exit (and it doesn't trigger any asserts in PhysX code), so that's why I never noticed these issues before....in another scene I tested with negative scale on colliding convex objects, I get no errors...and collisions are all normal, hull is normal, etc. 

I've now ensured the convex mesh scaling values I assign are always positive, and no longer get any crashing issues. So, this ultimately appears to be user error. Now that I've looked closer I do see the following line in the PxMeshScale header: 

    Only positive scale values are supported for PxConvexMeshGeometry

...but it might also be good to get a PX_ASSERT somewhere warning users that negative scale values on convex hulls are invalid, since they may not immediately cause any issues in the sim so it's harder to track down the cause when they do occur.

closed 2023-04-04 by tysoni

vreutskyy, 2023-04-04:
Yes, it definitely had to show an error when the invalid value was set. Thanks again for the report.