NVIDIA-Omniverse/PhysX #131: Issue running generate_projects.sh (ubuntu)

Kai-Indiemade, 2023-04-30:
Version: PhysX v5.1.0 on Ubuntu 22.04

Hi there, when running generate_projects.sh I get this error:

```
CMake Error at /home/laptop001/PhysX/physx/snippets/compiler/cmake/linux/SnippetRender.cmake:32 (find_package):

Could not find a package configuration file provided by "OpenGL" with any of the following names:

OpenGLConfig.cmake
opengl-config.cmake

Add the installation prefix of "OpenGL" to CMAKE_PREFIX_PATH or set

"OpenGL_DIR" to a directory containing one of the above files. If "OpenGL" provides a separate development package or SDK, be sure it has been installed.

Call Stack (most recent call first):
/home/laptop001/PhysX/physx/snippets/compiler/cmake/SnippetRender.cmake:32 (include)
/home/laptop001/PhysX/physx/snippets/compiler/cmake/CMakeLists.txt:111 (INCLUDE)
```

I do have gl packages(Such as libgl-mesa-dev, mesa-common-dev) installed on my system, as I have successfully used them in other projects.

Kai-Indiemade, 2023-05-01:
So I managed to get it working using the clang preset instead of gcc. 

If I set CMAKE_PREFIX_PATH to the folder containing opengl-config.cmake then "generate_proejcts.sh" worked correctly with gcc, but when trying to run the make file it said the file format of libGL.so was incorrect.

I'm good for now since I got it to work with clang, but you may want to still investigate the process with the gcc preset.

preist-nvidia, 2023-05-02:
@ayoub-belarbi can you please have a look at the gcc issue?

ayoub-belarbi, 2023-05-02:
Hello @Kai-Indiemade,
Thanks for reporting this issue. First question: What's the used/target architecture? is it x86_64 or aarch64? You mentioned that you changed from gcc to clang, which preset file is that exactly?
Thanks

Kai-Indiemade, 2023-05-02:
Hello @ayoub-belarbi , the target architecture that used gcc was aarch64.(linux-aarch64.xml)

The preset that used clang was linux.xml.