NVIDIA-Omniverse/PhysX #143: Error while generating projects

v71, 2023-05-31:
After downloading and running the generatre_projects.bat I get this error

Creating directory C:\packman-repo
Fetching python@3.7.9-windows-x86_64.cab ...
Downloading from bootstrap.packman.nvidia.com ...
Unpacking Python interpreter ...
Fetching packman-common@6.33.2.zip ...
Downloading from bootstrap.packman.nvidia.com ...
Unpacking ...
Processing project file 'C:\packman-repo\packman-common\6.33.2\deps.packman.xml'
Package '7za' at version '16.02.4' is missing from local storage.
Downloading from http://bootstrap.packman.nvidia.com/7za%4016.02.4.zip (4.33 MiB)
100.00% (speed 4.32 MiB/s)
Total of 1.00 seconds
Unzipping: 7za@16.02.4.zip (9.87 MiB)
100.00% (speed 90.1 MiB/s)
Total of 0.11 seconds
Package successfully installed to C:\packman-repo\chk\7za\16.02.4
  File "C:/Users/Vincenzo", line 1
    [1E40:0114][2023-05-24T17:01:31]i001: Burn v3.10.4.4718, Windows v10.0 (Build 19045: Service Pack 0), path: C:\WINDOWS\Temp\{6B707990-D955-4EF1-BCBC-78CDEDDE1213}\.cr\VC_redist.x64.exe
         ^
SyntaxError: invalid syntax
Hit ENTER to continue...

### Library and Version

### Operating System

### Steps to Trigger Behavior

 1.
 2.
 3.

### Code Snippet to Reproduce Behavior

```
<your-snippet>
```

### Expected Behavior

### Actual Behavior

ayoub-belarbi, 2023-06-21:
Hello @v71,
I really tried to reproduce this issue but couldn't. It seems something really specific to your system. Since I don't have access to your environment nor I can reproduce. I will have to close this issue.
Feel free to open it again if you have more information on how I can reproduce the issue.
Thank you

zzxzzk115, 2024-02-14:
Hi, I reproduce this issue because my user name contains a space! Please consider the path may contain spaces.

preist-nvidia, 2024-02-15:
@ayoub-belarbi can you please check the spaces-in-path case? Merci! Internal tracking: PX-4818

ayoub-belarbi, 2024-02-15:
Hi @zzxzzk115,

Which version of PhysX are you trying? The white space in the path issue has been fixed for a while now and hasn't been reported since.
Also, I tried to reproduce with white spaces in the path and it doesn't fail for me.

Thanks

zzxzzk115, 2024-02-15:
Hi @ayoub-belarbi,

It's 5.1.2 (104.1) both the official release and the O3DE fork release. I found this issue closed but has not been reproduced. I mainly want developers using version 5.1.2 to be aware of the issue with system usernames having spaces in them. The current workaround is to move the project folder to a directory without spaces. The current version has resolved this problem, and I appreciate your efforts.

Cheers

ayoub-belarbi, 2024-02-15:
Hi @zzxzzk115,

Version 5.1.2 is quite old, and we've fixed the problem with spaces in later updates. Specifically version 5.2.0: https://github.com/NVIDIA-Omniverse/PhysX/blob/main/physx/CHANGELOG.md?plain=1#L287

We only handle the main PhysX repo, so for anything related to the O3DE fork, it's best to get in touch with their maintainers. It's always a good idea to upgrade to the latest version when you can, to avoid bugs like this and get all the new features. If you're stuck on an older version, you might need to apply the fixes yourself.

Thanks,
Ayoub

closed 2024-02-15 by ayoub-belarbi