NVIDIA-Omniverse/PhysX #147: Possible thread safety issue with interaction handling

bdumesnil, 2023-06-03:
Hi

We are having an issue with corrupted mActiveInteractionCount in Sc::Scene ( values < 0 or incoherent with Interaction's id ) , after inquiring the issue we think we found two similar thread safety issues with Interaction handling.

### Library and Version

PhysX v5.1.0

### Operating System

Windows 10 / Windows 11

### Code to Reproduce Behavior

### Thread safety issue 1

In `Sc::Scene::postIslandGen`, _mSetEdgesConnectedTask_ and _mNPhaseCore->processTriggerInteractions_ can be called in concurrent tasks, 
https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScScene.cpp#L3695-L3704

#### Task 1
_mSetEdgesConnectedTask_ will call `Sc::Scene::setEdgesConnected` wich in turn call `wakeObjectsUp`
https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScScene.cpp#L3458-L3475
`wakeObjectsUp` may activate actors and their Interactions, wich will end up calling [`Sc::Scene::notifyInteractionActivated`](https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScScene.cpp#LL1536C17-L1536C43)
Heres is a possible callstack :
```
 physx::Sc::Scene::setEdgesConnected(physx::PxBaseTask * __formal)
	physx::Sc::Scene::wakeObjectsUp(unsigned int infoFlag)
		physx::Sc::ActorSim::setActive(bool active, unsigned int infoFlag)
			physx::Sc::BodySim::activate()
				physx::Sc::activateInteractions(physx::Sc::ActorSim & actorSim)
					physx::Sc::Scene::notifyInteractionActivated(physx::Sc::Interaction * interaction)
```
#### Task 2
_mNPhaseCore->processTriggerInteractions_ calls `NPhaseCore::mergeProcessedTriggerInteractions` wich will call [`Sc::Scene::notifyInteractionDeactivated`](https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScScene.cpp#L1553)

https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScNPhaseCore.cpp#L1632-L1647

#### Result
Unfortunately it does not seem to be valid to call `Sc::Scene::notifyInteractionActivated` and `Sc::Scene::notifyInteractionDeactivated` concurrently causing corrupted interactions count and indices.

A possible fix would be to modify `Sc::Scene::postIslandGen` to add a dependency between _mSetEdgesConnectedTask_ and _mNPhaseCore->processTriggerInteractions_

```
void Sc::Scene::postIslandGen(PxBaseTask* continuationTask)
{
	PX_PROFILE_ZONE("Sim.postIslandGen", getContextId());

	mSetEdgesConnectedTask.setContinuation(continuationTask);

	// - Performs collision detection for trigger interactions
	mNPhaseCore->processTriggerInteractions( &mSetEdgesConnectedTask );
	mSetEdgesConnectedTask.removeReference();
}
```

### Thread safety issue 2
A similar issue is located in `Sc::Scene::processLostContacts2`, `Sc::Scene::lostTouchReports` and `Sc::Scene::unregisterInteractions` will be called concurrently,
https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScScene.cpp#L3948-L3955

#### Task 1
`Sc::Scene::lostTouchReports` will wake actors on lost touch, wich will end up calling `Sc::Scene::notifyInteractionActivated`
https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScScene.cpp#L3870-L3894
Here is a possible callstack : 
```
physx::Sc::Scene::lostTouchReports(physx::PxBaseTask * __formal)
	physx::Sc::NPhaseCore::lostTouchReports(physx::Sc::ShapeInteraction * si, unsigned int flags, physx::Sc::ElementSim * removedElement, const unsigned int ccdPass, physx::PxsContactManagerOutputIterator & outputs)
		physx::Sc::BodySim::internalWakeUp(float wakeCounterValue)
			physx::Sc::BodySim::internalWakeUpBase(float)
				physx::Sc::ActorSim::setActive(bool active, unsigned int infoFlag)
					physx::Sc::BodySim::activate()
						physx::Sc::activateInteractions(physx::Sc::ActorSim & actorSim)
							physx::Sc::Scene::notifyInteractionActivated(physx::Sc::Interaction * interaction)
```

#### Task 2
`Sc::Scene::unregisterInteractions` will call `Sc::Scene::unregisterInteraction` .
https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScScene.cpp#L3896-L3919

#### Result
Similarly to the first issue `Sc::Scene::notifyInteractionActivated` and `Sc::Scene::unregisterInteraction` are not safe to be called concurrently.

The possible fix in `Sc::Scene::processLostContacts2` is to add a dependency between `Sc::Scene::lostTouchReports` and `Sc::Scene::unregisterInteractions` tasks : 
Before :
https://github.com/NVIDIA-Omniverse/PhysX/blob/93b6c250e2141f8e87543bc8bb763046f26ffbc5/physx/source/simulationcontroller/src/ScScene.cpp#L3948-L3955

After :

```
void Sc::Scene::processLostContacts2(PxBaseTask* continuation)
{
	mDestroyManagersTask.setContinuation(continuation);
	mLostTouchReportsTask.setContinuation( &mDestroyManagersTask );

	mUnregisterInteractionsTask.setContinuation( &mLostTouchReportsTask );
	mUnregisterInteractionsTask.removeReference();
	mLostTouchReportsTask.removeReference();
```

msauter-nvidia, 2023-06-07:
Hi Benoit,
Thanks a lot for this detailed bug report. We will look into it.

msauter-nvidia, 2023-06-08:
Hello Benoit,
To follow up on your report: "issue 1" looks not thread safe indeed. For "issue 2" I can not yet see the clash though. From what I see:

lostTouchReports => Sc::activateInteractions()
Sc::Scene::notifyInteractionActivated() will only get called for pairs that are not eOVERLAP nor eMARKER (for example trigger pairs)

Sc::Scene::unregisterInteractions() on the other hand only runs for eOVERLAP or eMARKER pairs

as such, it does looks safe since concurrent activation happens for different types of pairs only. I seem to be missing something. What was the object/scene setup where you saw "issue 2" trigger unsafe concurrent access?

bdumesnil, 2023-06-08:
Hi Michael,

Indeed issue 2 was a false positive on my side, my debug code was over-conservative and detected changes between interactions of different types.

msauter-nvidia, 2023-06-08:
Thanks for the confirmation. Fixing "issue 1" on our side will likely be postponed to 5.3.0. Please use your proposed workaround in the meantime.

closed 2023-06-08 by bdumesnil