NVIDIA-Omniverse/PhysX #153: Contact target velocity not zeroed out?

kermado, 2023-06-15:
It appears that calling `setTargetVelocity()` on a contact point can affect other contact points in subsequent time-steps. Perhaps the contact velocity is not zeroed out every time-step. This only appears to happen when using the CPU dispatcher rather than the GPU/CUDA dispatcher.

[image]

### Expected Behavior

The boxes below the conveyor resting on the floor should not move regardless of whether a box is moving on the conveyor.

### Actual Behavior

Some of the boxes below the conveyor move when a box is moving on the conveyor.

### Library and Version

PhysX v5.1.3

### Operating System

Windows 11

### Steps to Trigger Behavior

Run code snippet provided and press space to create a box.

### Code Snippet to Reproduce Behavior

The snippet provided below is a simple modification of the SnippetHelloWorld.cpp which reproduces the behavior.

```
#include <ctype.h>
#include "PxPhysicsAPI.h"
#include "../snippetcommon/SnippetPrint.h"
#include "../snippetcommon/SnippetPVD.h"
#include "../snippetutils/SnippetUtils.h"

using namespace physx;

static PxDefaultAllocator		gAllocator;
static PxDefaultErrorCallback	gErrorCallback;
static PxFoundation*			gFoundation = NULL;
static PxPhysics*				gPhysics	= NULL;
static PxDefaultCpuDispatcher*	gDispatcher = NULL;
static PxScene*					gScene		= NULL;
static PxMaterial*				gMaterial	= NULL;
static PxPvd*					gPvd        = NULL;

float conveyorSpeed = 0.5F;
PxRigidDynamic* conveyor = nullptr;

class ContactModifyCallback : public PxContactModifyCallback
{
public:
	void onContactModify(PxContactModifyPair* const pairs, PxU32 count)
	{
		for (PxU32 i = 0; i < count; ++i)
		{
			PxContactModifyPair& pair = pairs[i];

			// Conveyor.
			if (pair.actor[0] == conveyor || pair.actor[1] == conveyor)
			{
				for (PxU32 j = 0; j < pair.contacts.size(); ++j)
				{
					pair.contacts.setStaticFriction(j, 0.35F);
					pair.contacts.setDynamicFriction(j, 0.35F);
					pair.contacts.setNormal(j, PxVec3(0.0, 1.0, 0.0));
					pair.contacts.setTargetVelocity(j, PxVec3((pair.actor[0] == conveyor) ? conveyorSpeed : -conveyorSpeed, 0.0, 0.0));
				}
			}
			else
			{
				for (PxU32 j = 0; j < pair.contacts.size(); ++j)
				{
					pair.contacts.setStaticFriction(j, 0.35F);
					pair.contacts.setDynamicFriction(j, 0.35F);
					//pair.contacts.setTargetVelocity(j, PxVec3(0.0));
				}
			}
		}
	}
};

static PxFilterFlags ContactReportFilterShader(PxFilterObjectAttributes attributes0, PxFilterData filterData0,
	PxFilterObjectAttributes attributes1, PxFilterData filterData1,
	PxPairFlags& pairFlags, const void* constantBlock, PxU32 constantBlockSize)
{
	PX_UNUSED(attributes0);
	PX_UNUSED(attributes1);
	PX_UNUSED(filterData0);
	PX_UNUSED(filterData1);
	PX_UNUSED(constantBlockSize);
	PX_UNUSED(constantBlock);

	pairFlags = PxPairFlag::eDETECT_DISCRETE_CONTACT | PxPairFlag::eMODIFY_CONTACTS | PxPairFlag::eSOLVE_CONTACT;
	return PxFilterFlag::eDEFAULT;
}

static PxRigidDynamic* createBox(const PxVec3& position, const PxVec3& dimensions)
{
	PxTransform transform(position, PxQuat(PxIdentity));
	PxBoxGeometry geometry(dimensions * 0.5F);
	PxRigidDynamic* dynamic = PxCreateDynamic(*gPhysics, transform, geometry, *gMaterial, 10.0f);
	gScene->addActor(*dynamic);
	return dynamic;
}

static PxRigidDynamic* createLoad(const PxVec3& position, const PxVec3& dimensions)
{
	PxRigidDynamic* load = createBox(position, dimensions);
	load->setSolverIterationCounts(8, 2);
	load->setSleepThreshold(0.0);
	PxRigidBodyExt::setMassAndUpdateInertia(*load, 1.0);
	return load;
}

void initPhysics(bool interactive)
{
	PX_UNUSED(interactive);

	gFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gAllocator, gErrorCallback);

	gPvd = PxCreatePvd(*gFoundation);
	PxPvdTransport* transport = PxDefaultPvdSocketTransportCreate(PVD_HOST, 5425, 10);
	gPvd->connect(*transport,PxPvdInstrumentationFlag::eALL);

	gPhysics = PxCreatePhysics(PX_PHYSICS_VERSION, *gFoundation, PxTolerancesScale(),true,gPvd);

	PxSceneDesc sceneDesc(gPhysics->getTolerancesScale());
	sceneDesc.cudaContextManager = nullptr;
	gDispatcher = PxDefaultCpuDispatcherCreate(2);
	sceneDesc.cpuDispatcher = gDispatcher;
	sceneDesc.filterShader = PxDefaultSimulationFilterShader;
	sceneDesc.solverType = PxSolverType::ePGS;
	sceneDesc.filterShader = ContactReportFilterShader;
	sceneDesc.flags |= PxSceneFlag::eENABLE_GPU_DYNAMICS;
	sceneDesc.flags |= PxSceneFlag::eENABLE_PCM;
	sceneDesc.flags |= PxSceneFlag::eENABLE_FRICTION_EVERY_ITERATION;
	sceneDesc.broadPhaseType = PxBroadPhaseType::eABP;
	sceneDesc.contactModifyCallback = new ContactModifyCallback();
	sceneDesc.gravity = PxVec3(0.0f, -9.81f, 0.0f);
	
	gScene = gPhysics->createScene(sceneDesc);

	PxPvdSceneClient* pvdClient = gScene->getScenePvdClient();
	if(pvdClient)
	{
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONSTRAINTS, true);
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONTACTS, true);
		pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_SCENEQUERIES, true);
	}
	gMaterial = gPhysics->createMaterial(0.0F, 0.0F, 0.0F);
	gMaterial->setFlags(PxMaterialFlag::eIMPROVED_PATCH_FRICTION);

	PxRigidStatic* groundPlane = PxCreatePlane(*gPhysics, PxPlane(0,1,0,0), *gMaterial);
	gScene->addActor(*groundPlane);

	conveyor = createBox(PxVec3(0.0, 1.0, 0.0), PxVec3(2.0, 0.1, 0.3));
	conveyor->setRigidBodyFlag(PxRigidBodyFlag::eKINEMATIC, true);

	createLoad(PxVec3(-0.4, 0.05, 0.0), PxVec3(0.1, 0.1, 0.1));
	createLoad(PxVec3(-0.2, 0.05, 0.0), PxVec3(0.1, 0.1, 0.1));
	createLoad(PxVec3(0.0, 0.05, 0.0), PxVec3(0.1, 0.1, 0.1));
	createLoad(PxVec3(0.2, 0.05, 0.0), PxVec3(0.1, 0.1, 0.1));
	createLoad(PxVec3(0.4, 0.05, 0.0), PxVec3(0.1, 0.1, 0.1));
}

void stepPhysics(bool /*interactive*/)
{
	gScene->simulate(1.0f/100.0f);
	gScene->fetchResults(true);
}
	
void cleanupPhysics(bool /*interactive*/)
{
	PX_RELEASE(gScene);
	PX_RELEASE(gDispatcher);
	PX_RELEASE(gPhysics);
	if(gPvd)
	{
		PxPvdTransport* transport = gPvd->getTransport();
		gPvd->release();	gPvd = NULL;
		PX_RELEASE(transport);
	}
	PX_RELEASE(gFoundation);
	
	printf("SnippetHelloWorld done.\n");
}

void keyPress(unsigned char key, const PxTransform& camera)
{
	PX_UNUSED(camera);

	switch(toupper(key))
	{
	case ' ':	createLoad(PxVec3(0.4, 1.3, 0.0), PxVec3(0.3, 0.2, 0.2));	break;
	}
}

int snippetMain(int, const char*const*)
{
#ifdef RENDER_SNIPPET
	extern void renderLoop();
	renderLoop();
#else
	static const PxU32 frameCount = 100;
	initPhysics(false);
	for(PxU32 i=0; i<frameCount; i++)
		stepPhysics(false);
	cleanupPhysics(false);
#endif

	return 0;
}
```

jcarius-nv, 2023-07-31:
Thank you for reporting the issue and providing a repro. We identified the underlying problem, which will be fixed in an upcoming release. In the meantime, you may apply the following patch to your codebase:

In [DyContactPrep4.cpp:847](https://github.com/NVIDIA-Omniverse/PhysX/blob/8f6cf9cfe89f2bedafbf7788a72cb04f11c31e1f/physx/source/lowleveldynamics/src/DyContactPrep4.cpp#L847) replace as follows:

 					Vec4V targetVel0 = V4LoadA(contactIndex0 == 0xFFFF ? &contactBase0->targetVel.x : &descs[0].contacts[contactIndex0].targetVel.x);
	-				Vec4V targetVel1 = V4LoadA(contactIndex1 == 0xFFFF ? &contactBase0->targetVel.x : &descs[1].contacts[contactIndex1].targetVel.x);
	-				Vec4V targetVel2 = V4LoadA(contactIndex2 == 0xFFFF ? &contactBase0->targetVel.x : &descs[2].contacts[contactIndex2].targetVel.x);
	-				Vec4V targetVel3 = V4LoadA(contactIndex3 == 0xFFFF ? &contactBase0->targetVel.x : &descs[3].contacts[contactIndex3].targetVel.x);
	+				Vec4V targetVel1 = V4LoadA(contactIndex1 == 0xFFFF ? &contactBase1->targetVel.x : &descs[1].contacts[contactIndex1].targetVel.x);
	+				Vec4V targetVel2 = V4LoadA(contactIndex2 == 0xFFFF ? &contactBase2->targetVel.x : &descs[2].contacts[contactIndex2].targetVel.x);
	+				Vec4V targetVel3 = V4LoadA(contactIndex3 == 0xFFFF ? &contactBase3->targetVel.x : &descs[3].contacts[contactIndex3].targetVel.x);

closed 2023-07-31 by jcarius-nv

hazidh, 2025-03-19:
Hello,sir.I met a problem in the use of the `contacts.setTargetVelocity` this method to simulate a circular conveyor belt, I add a tangential velocity for each contact, but the revolution of the object will be more and more big, this how to solve, looking forward to your reply.

https://github.com/user-attachments/assets/ceed8474-8466-4996-9b56-087fa066091a

```
class ConveyorBeltCallback : public PxContactModifyCallback
{
    void onContactModify(PxContactModifyPair* const pairs, PxU32 count) override
    {
        for (PxU32 i = 0; i < count; i++)
        {
            PxContactSet& contacts = pairs[i].contacts;
            for (PxU32 j = 0; j < contacts.size(); j++)
            {
                PxVec3 center(0, 0, 0);  // 传送带中心
                PxVec3 tangentDirection = (point - center).cross(PxVec3(0, 1, 0)); 
                tangentDirection.normalize();
                PxVec3 conveyorVelocity = tangentDirection * 3.0f; 
                contacts.setTargetVelocity(j,conveyorVelocity);
            }
        }
    }
};
```