NVIDIA-Omniverse/PhysX #156: Contact generated despite absolute seperation value higher than contact offset

alektron, 2023-06-29:
### Library and Version
release/104.2

### Operating System
Windows 10

### Description
For Capsule-Mesh contact generation, contacts can be generated whose seperation value is higher than the contact offset.
In some scenarios (image below) with relatively thin objects this can lead to contacts being generated on the opposite side of the colliding objects.

[image]

### Proposed fix
[Pull request soon...]
The contacOffset check compares against the t/distance value of intersectRayTriangle.
This value can however be negative, so the comparison `t < radius + contactDistance` can be true despite the absolute value of `t` being higher than `radius + contactDistance`

vreutskyy, 2023-07-04:
Hello, alektron
The negative distance is considered a contact despite of it's magnitude. Like if you'd push this capsule inside a plane or a solid box. No matter how deep it would be, it'd be still a contact. So, it's not a bug, but still, thank you for the report.

alektron, 2023-07-06:
That does make sense.
However I'd argue that the generated contact is still caused by some bug, even if my guess at the cause wasn't correct.
The behaviour of the dynamic rigid body in the scene (the capsule) was very much unusable before my 'fix' due to these contacts.
With the capsule jumping around and eventually getting shot away as soon as it entered the hole of the 'chain' visible in the above image.

As previously stated I encountered the bug with PhysX 4.1.
Since the code, that I assumed was the problem, is present in 4.1 AND 5.0 I figured it'd make more sense to post it here.
I will try to update my project to PhysX 5 and see if the problem still persists and if it does I can try to create a minimal example.

closed 2023-10-27 by preist-nvidia