NVIDIA-Omniverse/PhysX #161: Spring/damper doing nothing for articulations

mlp1802, 2023-07-06:
When trying to drive my articulation joints, only "max_force" affects my articulation moving speed. Spring/damper can be zero or a billion, it makes no difference. What could be wrong? Physx 5.1.3..(used from rust but that shouldn't matter)
```rust
pub fn set_articulation_joint_reduced_target(
    joint: PtrArticulationJointReducedCoordinate,
    axis: PxArticulationAxis,
    max_force: f32,
    spring: f32,
    damper: f32,
    target: f32,
) {
    unsafe {
        let drive = PS::PxArticulationDrive_new_1(
            spring,
            damper,
            max_force,
            PS::PxArticulationDriveType::Target,
        );

        PS::PxArticulationJointReducedCoordinate_setDriveParams_mut(joint, axis, &drive);
        PS::PxArticulationJointReducedCoordinate_setDriveTarget_mut(joint, axis, target, true);
        }
}
```

preist-nvidia, 2023-07-10:
I am not sure - but I am sure that the drives work - please compare your setup to one of the snippets, e.g: https://github.com/NVIDIA-Omniverse/PhysX/blob/main/physx/snippets/snippetfixedtendon/SnippetFixedTendon.cpp#L126

This drive works.

mlp1802, 2023-07-10:
Differences from my code is:
1) My code uses drive type TARGET
2) I set the drive params before every step (in the snippet it's set once)
3) Minor differences to how scene is created.
I'll try to make a minimal example to see if I can pinpoint it furter.

mlp1802, 2023-07-10:
Oops I see the error. Setting drive params is not allowed while the simulation is running. But I need to set spring/damper dynamically or I'm just having the same speed of animation constantly, how can that be done?

preist-nvidia, 2023-07-10:
The target does nothing but hardcode to a fixed very large position gain and zero velocity gain: https://github.com/NVIDIA-Omniverse/PhysX/blob/main/physx/source/lowleveldynamics/src/DyFeatherstoneArticulation.cpp#L2644

And then set the drive to force mode. FYI we will deprecate this mode in 5.3.0.

preist-nvidia, 2023-07-10:
> Oops I see the error. Setting drive params is not allowed while the simulation is running. But I need to set spring/damper dynamically or I'm just having the same speed of animation constantly, how can that be done?

Instead of the gains, you can also set the drive target position/velocity. Here for position: https://nvidia-omniverse.github.io/PhysX/physx/5.2.1/_build/physx/latest/class_px_articulation_joint_reduced_coordinate.html?highlight=setdrivetarget#_CPPv4N36PxArticulationJointReducedCoordinate14setDriveTargetEN18PxArticulationAxis4EnumEK6PxRealb

preist-nvidia, 2023-07-10:
The snippet updates the drive target, FWIW.

mlp1802, 2023-07-10:
Yes I do update in between simulation steps..how else can you update it? Maybe during some callback or something? (I don't use any callbacks)

mlp1802, 2023-07-10:
Ah ok so you want me to use force mode? I'm not sure, how can you drive a joint towards an orientation like that?

mlp1802, 2023-07-10:
> Instead of the gains, you can also set the drive target position/velocity. Here for position: [...]

Am I not already doing that ?
`S::PxArticulationJointReducedCoordinate_setDriveTarget_mut(joint, axis, target, true);`

mlp1802, 2023-07-10:
Solved! I was under the impression that FORCE joint drive type would make the         PS::PxArticulationJointReducedCoordinate_setDriveTarget_mut(joint, axis, target, true); drive towards a "FORCE" target. But now I understand that it's just a normal position. 
So using the type FORCE works, spring and damper and all. Could maybe have been explained a little better in the docs, considering the complexity. But when the end is good everything is good, thanks!

closed 2023-07-10 by mlp1802