NVIDIA-Omniverse/PhysX #164: (Reduced coordinate) articulations keeps crashing.

mlp1802, 2023-07-10:
Sorry to bother you again with this..I've just tried everything!
I have two articulations in the scene...when they collide, 3 out of ten times the game crashes.
Sometimes it's this error:
Assertion failed: PxAbs(float(1.0) - unitAxis.magnitude()) < float(1e-3)
Sometimes this:
Asertion failed: motionAccelerations[linkID].isFinite()
Sometimes this:
Assertion failed: PxIsFinite(stZ)

It seems to happen when they reach a certain speed, but not always.

Ive tried setting center of mass limits to various values
```
setMaxCOMLinearVelocity_mut(ptr, 100.0);
setMaxCOMAngularVelocity_mut(ptr, 100.0);
```
Tried setting max velocity on rigid bodies.

And scaled the ragdoll and it's weight  from dimensions between everything between 0.1 and 3.0 and weight from anything between 1 and 1000. Nothing helps

preist-nvidia, 2023-07-11:
Hi Mikkel - this would indicate something going unstable triggering the finite checks. Do you see something exploding? Do you have reasonable mass ratios (say < 100) between all the links/objects in the scene?

Another thing to try is to add velocity limits to the joints to also clamp in the reduced coordinates. https://nvidia-omniverse.github.io/PhysX/physx/5.2.1/_build/physx/latest/class_px_articulation_joint_reduced_coordinate.html?highlight=setdrivetarget#_CPPv4N36PxArticulationJointReducedCoordinate19setMaxJointVelocityEK6PxReal

mlp1802, 2023-07-11:
Yes ok I will try velocity clamps..but other than that I see no reason why it would constantly explode, the max ratio between joints is around one to 4. Anyway I see no difference between the two solvers really so it's not a big problem, will fiddle around with it as the game progress and post the result here once I get it working again

preist-nvidia, 2023-07-11:
Ok please let me know how it is going. We really should not have a big difference between TGS and PGS. Just in case, are you using joint drives and are you running high velocity iterations?

mlp1802, 2023-07-11:
> using joint drives and are you running high velocity iterations

Maybe, I'm not sure,  it's a boxing game, so some speed is involved. I tried setting max joint velocity as well, didn't do anything (other than slowing everything down). I'm sure it will be solved somehow.

> And are not using the latest SDK?

There's a commit on github that simply says "Upgrade to PhysX 5 " 
https://github.com/EmbarkStudios/physx-rs/pull/183
I'm not sure if they use the latest and greatest

mlp1802, 2023-07-11:
Thanks a lot! It works nicely now (with the other solver) though I still don't understand how reduced coordinate is better. Seems to me the simulation is worse (than max coord) but that might just be a matter of tweaking. It's hard to post single snippets as there are a few abstracts here and there, but I could give you access to the repo if it's needed. Have to warn you there's like 100k lines of code or so.

preist-nvidia, 2023-07-11:
Ok good! No need to give me access - we really only can address repros as snippets due to limited support engineering time we have.

closed 2023-07-11 by preist-nvidia

mlp1802, 2023-07-11:
Right, I'll post if there's a solution.

mlp1802, 2023-07-13:
Solved I believe. I was using a spherical joint as a hinge joint, one dof. I read that they should have two or three. Anyway, after using a revolute joint instead, the application stopped crashing. 
Despite from crashing, the joint also seemed to drift slowly out of position, though it would look ok in the beginning of the simulation.

mlp1802, 2023-07-13:
And TGS solver now works without that crash.

preist-nvidia, 2023-07-14:
Ok thank you for the report - yes, a 1dof rotational joint should be a revolute. The spherical joint should also work, but we are aware of some issues with it.