NVIDIA-Omniverse/PhysX #169: Debug visualization is one frame behind

rlidwka, 2023-07-16:
I ran `scene->simulate()`, then compared actor position (from `dynamic->getGlobalPose()`) with debug visualization (from `scene->getRenderBuffer()`).

 - expected behavior: after each simulation step, I expect actor position to match debug visualization in render buffer
 - actual behavior: debug visualization is lagging 1 frame behind

In the following example, I create a free-falling sphere, then print out it's position and actor axes:

```c++
#include <ctype.h>
#include "PxPhysicsAPI.h"

using namespace physx;

PxDefaultAllocator		gAllocator;
PxDefaultErrorCallback	gErrorCallback;

int main()
{
	// standard boilerplate
	auto gFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gAllocator, gErrorCallback);
	auto gPhysics = PxCreatePhysics(PX_PHYSICS_VERSION, *gFoundation, PxTolerancesScale());
	PxSceneDesc sceneDesc(gPhysics->getTolerancesScale());
	sceneDesc.gravity = PxVec3(0.0f, -9.81f, 0.0f);
	auto gDispatcher = PxDefaultCpuDispatcherCreate(2);
	sceneDesc.cpuDispatcher = gDispatcher;
	sceneDesc.filterShader = PxDefaultSimulationFilterShader;
	auto gScene = gPhysics->createScene(sceneDesc);

	// creating a free-falling sphere
	auto gMaterial = gPhysics->createMaterial(0.5f, 0.5f, 0.1f);
	PxRigidDynamic* dynamic = PxCreateDynamic(*gPhysics, PxTransform(PxVec3(0, 0, 0)), PxSphereGeometry(10), *gMaterial, 100.0f);
	dynamic->setAngularDamping(0.5f);
	gScene->addActor(*dynamic);

	gScene->setVisualizationParameter(PxVisualizationParameter::eSCALE, 1.0f);
	gScene->setVisualizationParameter(PxVisualizationParameter::eACTOR_AXES, 1.0f);

	for (auto i = 0; i < 10; i++) {
		gScene->simulate(1.0f / 60.0f);
		gScene->fetchResults(true);

		auto pose = dynamic->getGlobalPose();
		//printf("position: %f %f %f\n", pose.p.x, pose.p.y, pose.p.z);
		auto actor_position = pose.p.y;

		const PxRenderBuffer& rb = gScene->getRenderBuffer();
		float render_position = 0;
		for (PxU32 i = 0; i < rb.getNbLines(); i++)
		{
			const PxDebugLine& line = rb.getLines()[i];
			//printf("debug: %f %f %f, %f %f %f\n", line.pos0.x, line.pos0.y, line.pos0.z, line.pos1.x, line.pos1.y, line.pos1.z);
			render_position = line.pos0.y;
		}

		printf("%f %f\n", actor_position, render_position);
		//printf("----------------------\n");
	}
}
```

Here's the output:

```
-0.002725 0.000000
-0.008175 -0.002725
-0.016350 -0.008175
-0.027250 -0.016350
-0.040875 -0.027250
-0.057225 -0.040875
-0.076300 -0.057225
-0.098100 -0.076300
-0.122625 -0.098100
-0.149875 -0.122625
```

I expected to see the same numbers in each row.

Windows 11, PhysX 5.1.3.

preist-nvidia, 2023-07-17:
Hi Alex,

This is by design. The debug visualization uses the transforms at the beginning of the simulation step, not the end. This is because the contacts are generated using the pre-sim-step transforms, and the contact debug visualization would be off if we used the post-integration/solve transforms.

We don't state this explicitly in our docs, though. I'll add something to clarify this.

Philipp

rlidwka, 2023-07-17:
> This is by design. The debug visualization uses the transforms at the beginning of the simulation step, not the end.

Okay, thanks. I suspected that this was the case, but just wanted to confirm.

Please add a mention about it in the docs on Debug Visualization.

closed 2023-07-17 by rlidwka