NVIDIA-Omniverse/PhysX #176: MSVC C++20 deprecated include warning

BobbyAnguelov, 2023-08-07:
In trying to switch some code over to C++20, I'm getting this warning:

```
warning C4996: '_Header_ciso646': warning STL4036: <ciso646> is removed in C++20. You can define _SILENCE_CXX20_CISO646_REMOVED_WARNING or _SILENCE_ALL_CXX20_DEPRECATION_WARNINGS to suppress this warning.
```

For reference: https://en.cppreference.com/w/cpp/header/ciso646

This is coming from PxPreprocessor.h, ln 34.

One possible solution is:

```
#if __cplusplus <= 201703L
    #include <ciso646> // and, not, or
#endif
```

Thanks!

thebrandre, 2023-10-19:
Any news here?

We could also define `PX_GENERATE_META_DATA`, which deesn't seem to have any effect other than avoiding the include.
To me, it looks very much like legacy that no longer serves any purpose.
It would be great if someone could confirm this! Or even better: fix it ;-)

Thanks!

preist-nvidia, 2023-10-27:
Hi both - sorry for the late response. We only support the compiler settings as generated by our presets. For other build configurations please check/contribute to the O3DE fork: https://github.com/NVIDIA-Omniverse/PhysX#community-maintained-build-configuration-fork

thebrandre, 2023-10-28:
@preist-nvidia Just to be clear: you don't have to use any build configuration other than the default for that issue to occur. And it is *not* addressed by the O3DE fork.

The warning appears if the public API of PhysX is consumed by a project that uses C++20 features in Visual Studio 2022, which is built as an officially supported configuration! 
I guess it's in your best interest to keep up to date with the current compilers. So we were trying to make you aware of that ... and were not making any demands (as your response somehow suggests).
It would be nice if you could reopen the issue, as the issue still persists and it will affect an increasing number of users as time progresses.

preist-nvidia, 2023-10-30:
Hi @thebrandre - thank you for the clarification! Let me reopen and I'll check how to best proceed here.

preist-nvidia, 2023-11-01:
@thebrandre @BobbyAnguelov - we removed the offending header include and this change will be in the next release.

closed 2023-11-01 by preist-nvidia