NVIDIA-Omniverse/PhysX #182: PxTriangleMesh collision error

preist-nvidia, 2023-08-16:
### Discussed in https://github.com/NVIDIA-Omniverse/PhysX/discussions/181

<div type='discussions-op-text'>

<sup>Originally posted by **yananZZZZ** August 16, 2023</sup>
### triangle mesh collide with triangle mesh
Hi, 
As I tried to make two mesh shape collide with each others, I got error reported in the GPU dll:

`PxgCudaSolverCore.cpp (1702) : internal error : GPU mComputeBodiesAverageVelocitiesFunction fail to launch kernel!!`
`PxgCudaSolverCore.cpp (1681) : internal error : GPU solveStaticBlock fail to launch kernel!!`
`PxgCudaUtils.h (75) : internal error : SynchronizeStreams cuEventRecord failed with error 700`
`PxgCudaUtils.h (81) : internal error : SynchronizeStreams cuStreamWaitEvent failed with error 700`

As one of my actor are `PxRigidDynamic` I enable the sdf when cook my mesh shape, something like:
[image]

since GPU dll is not open sourced, and I have no idea how to debug this kind of issue as mesh vs mesh is a common case in our application, hope someone can help to give some advice here.
I also export the ovd file for this scene maybe it a bit helpful to reproduce the issue.
[image]

[MeshCollisionPVD.zip](https://github.com/NVIDIA-Omniverse/PhysX/files/12354444/MeshCollisionPVD.zip)

</div>

preist-nvidia, 2023-08-16:
Thank you for the report @yananZZZZ we will have a look and get back to you.

nvtw, 2023-08-16:
Hello, I had a look at the the ovd capture but I could not find something that is obviously incorrect. It might be worth to try a couple of things to see if it starts to work afterwards:

- Ensure that the sdf spacing leads to a reasonable sdf resolution. An estimate of the resolution will be boundingBoxDiagonal/sdfSpacing. I tested the scene in omniverse and it worked quite well with a target sdf resolution of 500. The options to sparsify (compress) are identical as in your scene.
- Maybe it is worth to try a kinematic rigid collider instead of a static collider. But a kinematic rigid would also require an sdf. But since both parts are identical, the sdf can be shared.

The file I used to experiment is attached. Maybe it helps to copy some settings.

[MeshCollisionTest2.zip](https://github.com/NVIDIA-Omniverse/PhysX/files/12360035/MeshCollisionTest2.zip)

nvtw, 2023-08-21:
Omniverse 104.x (I guess that's equivalent 2022.3.x) should use the PhysX SDK 5.1.x. Usually the SDKs minor version increases with every new major version of kit. So 2022.4 would probably use 5.2.x but I'm not perfectly well informed on that.
I can try to set up your scene in the PhysX SDK to see if I run into problems too.
About cooking: There is a way to create SDFs on the gpu but it has not made it into the documentation yet. I'll update the docs asap. In the meantime you can try to set it up without documentation as it is quite easy: In the PxSDFDesc structure, make sure the member sdfBuilder points to a valid instance of the gpu sdf builder
`sdfDesc.sdfBuilder = PxGetPhysicsGpu()->createSDFBuilder(cudaContextManager);`
Make sure you release the sdfBuilder instance after the cooking to avoid memory leeks since the memory is owned by the caller. You might need to inlcude #include "PxSDFBuilder.h"

nvtw, 2023-08-21:
You're right, the SDFBuilder seems to be in the 5.3 SDK version which is not out yet, sorry my fault. Omniverse also has caching built in, so if you compute the same sdf twice, it will take it from the cash after computing it once.

- SDF resolution is not an SDK parameter. Omniverse computes the spacing out of the resolution. So what omnivese does is it takes the bounding box of the mesh, divides the length longest bounding box axis by the resolution and sets the result as spacing. When designing the API, we thought SDF spacing would be a good parameter but we probably should have chosen the resolution.
- The face count in omniverse is unfortunately not directly visible to my knowledge. But there is a small pyhton script editor in omniverse where you can get your mesh from the stage and print the face count to the screen.
- Tolerance scale is used for hole filling when cooking meshes and other tolerances. I never had to adjust it and I'm not an expert when it comes to that parameter. I can only recommend to look it up in the documentation.
- The contact buffer size defines the maximum number of contacts that can get processed in a single physics step. Increasing it by a factor of 2 or 4 is good for sdfs since they tend to produce many contacts. For penetration issues, the root cause is more often a too big timestep or a too small contact offset. Especially thin meshes need to be handled with care when it comes to collisions, especially if they pick up high speed in e. g. free fall. The omniverse scene I shared before also has speculative CCD (continuous contact detection) on. This mechanism is not perfect and costs a bit of performance when objects move fast, but it helps to avoid missing contacts, especially if thin objects move quickly.

nvtw, 2023-08-21:
Do you have code you could share with me such that I can reproduce the issue in the SDK? Usually I tend to modify the SDF snippet such that it uses different meshes and different parameters since it is an easy way to get some basic rendering. Having some code from you would simplify the support.

nvtw, 2023-09-20:
Sorry for the long delay with answering. I now created a cpp file which you can use to overwrite the SnippetSDF.cpp. It expects the off-3d data file at the location C:\tmp. Feel free to adjust the folder name in the code. Collisions between the complex geometries work but I have not yet tried that base plate rectangle. Please note that SDFs work best on meshes with uniform scaling. Non uniform scaling is not recommended. If non uniform scaling is required, please scale the vertices non-uniformly before you feed them into the cooking pipeline.
[SnippetSDFwithCustomMesh.zip](https://github.com/NVIDIA-Omniverse/PhysX/files/12672761/SnippetSDFwithCustomMesh.zip)

preist-nvidia, 2023-12-11:
Closing for now, please reopen if we should pick this up again.