NVIDIA-Omniverse/PhysX #185: Leaks in Blast using NvBlastExtAuthoringFindAssetConnectingBonds

LaBouette, 2023-08-23:
### Discussed in https://github.com/NVIDIA-Omniverse/PhysX/discussions/183

<div type='discussions-op-text'>

<sup>Originally posted by **LaBouette** August 22, 2023</sup>
Hi, 

I have some memory leaks using the blast plugin, that comes from using the NvBlastExtAuthoringFindAssetConnectingBonds method.
If I go through the log I have for the leaks, I have a call to NvBlastExtAuthoringTransformCollisionHull in the method, line 527 of NvBlastExtAuthoring.cpp. It then points me to these allocations in NvBlastExtAuthoringTransformCollisionHull (line 174 to 176 of NvBlastExtAuthoring.cpp) :
```
ret->points = SAFE_ARRAY_NEW(NvcVec3, ret->pointsCount);
ret->indices = SAFE_ARRAY_NEW(uint32_t, ret->indicesCount);
ret->polygonData = SAFE_ARRAY_NEW(HullPolygon, ret->polygonDataCount);
```
The CollisionHulls allocated by this method are stored in hullToRelease array, which then deletes all these CollisionHulls (line 574) : 
```
for (CollisionHull* hull : hullsToRelease)
{
    delete hull;
}
```
I did implement my custom CollisionHullImpl, to go with my custom ConvexMeshBuilder, but the destructor of my CollisionHullImpl does release these arrays (and I don't have leaks problems when using my ConvexMeshBuilder) : 

```
CollisionHullImpl::~CollisionHullImpl()
{
    SAFE_ARRAY_DELETE(points);
    SAFE_ARRAY_DELETE(indices);
    SAFE_ARRAY_DELETE(polygonData);
}
```
It seems the base CollisionHull struct has no destructor at all. The leaks might come from this?</div>

LaBouette, 2023-10-27:
Hi,

Any news on this?
Were you able to reproduce it?

Thanks in advance.

nvidia-bgaldrikian, 2023-11-01:
Hi LaBouette,

I'm very sorry this has taken far too long.  We'll publish a version with these leaks fixed as soon as we can.  I'll update you with the progress!

Thank you for reporting the leaks,
Bryan Galdrikian

nvidia-bgaldrikian, 2023-11-14:
Hi,

Blast-SDK 5.0.3 has been pushed.  Please let me know if that works for you.

Bryan

LaBouette, 2023-11-15:
Hi,

That is perfect, thank you.

closed 2023-11-15 by LaBouette

nvidia-bgaldrikian, 2023-11-17:
My pleasure.