NVIDIA-Omniverse/PhysX #191: Particle Systems - Dynamic Coupling

steinraf, 2023-09-07:
### Library and Version
PhysX v5.1.0

### Operating System
Ubuntu 22.04.3 LTS

### Steps to Trigger Behavior
 1. Add code file directly or via simlink to the `omni/physxdemos/scenes` folder
 2. Load Scene in Isaac Sim 2022.2.0 under `Window -> Simulation -> Demo Scenes -> Deformables -> Particle Sampler Extension`
 3. Press play. After 3 seconds one of the cubes is lifted and dropped, leading to cubes suspended in other particle systems jumping up.

### Code Snippet to Reproduce Behavior

```
from omni.physx.scripts import physicsUtils, particleUtils
from pxr import UsdGeom, Sdf, Gf, PhysxSchema, UsdPhysics
import omni.usd
import omni.physxdemos as demo
from omni.physxdemos.scenes.ParticleDemoBaseDemo import ParticleDemoBase

# Extension of ParticleSampler PhysX Demo
class ParticleSamplerDemoExtension(demo.Base, ParticleDemoBase):
    title = "Particle Sampler Extension"
    category = demo.Categories.DEFORMABLES
    short_description = "Extension of the Particle Sampler Demo"
    description = "Particle Systems are initialized in a grid with varying parameters"

    demo_camera = "/World/Camera"

    def create(
        self,
        stage,
    ):
        self.setup_base_scene(stage)

        default_prim_path = stage.GetDefaultPrim().GetPath()

        def create_system(
            stage,
            id,
            pos=Gf.Vec3f([0, 0, 0]),
            num_particles=32 * 32 * 32,
            ParticleContactOffset=2.0,
            Density=10.0,
            Friction=0.2,
            Damping=0.0,
            Viscosity=0.0,
            Cohesion=0.0,
            SurfaceTension=0.0,
            Drag=0.0,
            Lift=0.0,
            do_cube=True,
        ):
            # configure and create particle system
            particle_system_path = default_prim_path.AppendChild(f"particleSystem_{id}")
            particle_system = PhysxSchema.PhysxParticleSystem.Define(stage, particle_system_path)
            particle_system.CreateSimulationOwnerRel().SetTargets([self._scene.GetPath()])
            # The simulation determines the other offsets from the particle contact offset
            particle_system.CreateParticleContactOffsetAttr().Set(ParticleContactOffset)

            # create particle material and assign it to the system:
            particle_material_path = default_prim_path.AppendChild(f"particleMaterial_{id}")
            particleUtils.add_pbd_particle_material(
                stage,
                particle_material_path,
                density=Density,
                friction=Friction,
                damping=Damping,
                viscosity=Viscosity,
                cohesion=Cohesion,
                surface_tension=SurfaceTension,
                drag=Drag,
                lift=Lift,
            )

            physicsUtils.add_physics_material_to_prim(
                stage, stage.GetPrimAtPath(particle_system_path), particle_material_path
            )

            # create a cube mesh that shall be sampled:
            cube_mesh_path = Sdf.Path(omni.usd.get_stage_next_free_path(stage, f"/Cube", True))
            cube_resolution = 2  # resolution can be low because we'll sample the surface / volume only irrespective of the vertex count
            omni.kit.commands.execute(
                "CreateMeshPrimWithDefaultXform", prim_type="Cube", u_patches=cube_resolution, v_patches=cube_resolution
            )
            cube_mesh = UsdGeom.Mesh.Get(stage, cube_mesh_path)
            # physicsUtils.set_or_add_scale_op(cube_mesh, Gf.Vec3f(2.0, 2.0, 2.0))
            physicsUtils.set_or_add_translate_op(cube_mesh, Gf.Vec3f(0.0, 1 * 55.0, 0.0) + pos)

            # configure target particle set:
            particle_points_path = default_prim_path.AppendChild(f"sampledParticles_{id}")
            points = UsdGeom.Points.Define(stage, particle_points_path)
            # add render material:
            material_path = self.create_pbd_material(f"OmniPBR_{id}")
            omni.kit.commands.execute(
                "BindMaterialCommand", prim_path=particle_points_path, material_path=material_path, strength=None
            )
            particle_set_api = PhysxSchema.PhysxParticleSetAPI.Apply(points.GetPrim())
            particle_set_api.CreateParticleSystemRel().SetTargets([particle_system_path])

            particle_set_api.GetPrim().CreateAttribute("physxParticle:fluid", Sdf.ValueTypeNames.Bool).Set(True)

            particle_set_api.GetPrim().CreateAttribute("physxParticle:particleGroup", Sdf.ValueTypeNames.Float).Set(id)

            # physicsUtils.configure_particle_set()

            # compute particle sampler sampling distance
            # use particle fluid restoffset to determine sampler distance, using same formula as simulation, see
            # https://docs.omniverse.nvidia.com/prod_extensions/prod_extensions/ext_physics.html#offset-autocomputation
            fluid_rest_offset = 0.99 * 0.6 * ParticleContactOffset
            particle_sampler_distance = 2.0 * fluid_rest_offset

            # reference the particle set in the sampling api
            sampling_api = PhysxSchema.PhysxParticleSamplingAPI.Apply(cube_mesh.GetPrim())
            sampling_api.CreateParticlesRel().AddTarget(particle_points_path)
            sampling_api.CreateSamplingDistanceAttr().Set(particle_sampler_distance)
            sampling_api.CreateMaxSamplesAttr().Set(num_particles)
            sampling_api.CreateVolumeAttr().Set(True)

            # # Cube path
            cubeActorPath = f"/cubeActor_{id}"

            # Cube parameters like size, position, orientation
            size = 50.0
            position = Gf.Vec3f(0.0, 150.0, -10.0) + pos
            orientation = Gf.Quatf(1.0)
            color = Gf.Vec3f(71.0 / 255.0, 105.0 / 255.0, 1.0)

            # define the cube geom
            cubeGeom = UsdGeom.Cube.Define(stage, cubeActorPath)
            cubeGeom.CreateSizeAttr(size)
            cubeGeom.AddTranslateOp().Set(position)
            cubeGeom.AddOrientOp().Set(orientation)
            cubeGeom.AddScaleOp().Set(Gf.Vec3f(1.0))
            cubeGeom.CreateDisplayColorAttr().Set([color])

            # get the prim
            cubePrim = stage.GetPrimAtPath(cubeActorPath)

            # setup collision, so that the cube does collider
            UsdPhysics.CollisionAPI.Apply(cubePrim)
            UsdPhysics.RigidBodyAPI.Apply(cubePrim)
            UsdPhysics.MassAPI.Apply(cubePrim)

            # create catch box:
            box_path = f"box_{id}"

            self.create_particle_box_collider(
                default_prim_path.AppendChild(box_path),
                side_length=150.0,
                height=200.0,
                thickness=20.0,
                translate=Gf.Vec3f(0, -5, 0) + pos,
            )

            return cubePrim

        num_z, num_x = 4, 4

        for z in range(num_z):
            for x in range(num_x):
                id = num_x * z + x
                cube = create_system(
                    stage,
                    id,
                    Gf.Vec3f([(x - num_x / 2) * 200, 0, (z - num_z / 2) * 200]),
                    num_particles=32 ** 3,
                    ParticleContactOffset=2.0,
                    Density=200,
                    Friction=5 * z,
                    Damping=0.0,
                    Viscosity=100 * x,
                    Cohesion=100 * x,
                    SurfaceTension=0.0,
                    Drag=0.0,
                    Lift=0.0,
                )
                if id == 0:
                    self.lift_cube = cube

        self.t = 0
        self.do_lifting = True

    def update(self, stage, dt, viewport, physxIFace):
        if not omni.timeline.get_timeline_interface().is_playing():
            return

        self.t += dt

        if self.do_lifting and self.t > 3:
            self.do_lifting = False

            cube_geom = UsdGeom.Cube(self.lift_cube)

            for op in cube_geom.GetOrderedXformOps():
                if op.GetOpType() == UsdGeom.XformOp.TypeTranslate:
                    p = op.Get()
                    op.Set(Gf.Vec3f([p[0], 400, p[2]]))

```

### Expected Behavior
The particle systems with different Particle Group ID's should have no influence on each other.

### Actual Behavior

Disturbances in one particle systems affect the other particle systems. 
One example of this is shown by modifying the Particle Sampler Demo to have multiple particle systems, dropping a cube on one of the systems, and seeing how other cubes suspended in the other particle system receive an impulse launching them up.

sschirm, 2023-09-18:
Hello @steinraf , 
The concept of particle groups is a bit non-intuitive (it was originally borrowed from FleX). The group Id is part if the particle phase and it's meant to identify a set of particles belonging to the same "phase". Two particles of the **same** group don't interact with each other, unless the phase also specifies self-collision. Two particle with **different** groups actually do interact with each other. The only reasonable way to create N non-interacting pools of particles, is to create one particle system for each pool.

Cheers, 
Simon

closed 2023-10-27 by preist-nvidia

steinraf, 2023-11-22:
I was in contact with you via mail, but since this issue here was closed and we didn't chat any further I thought I would comment again to see if there was any progress!

sschirm, 2023-11-22:
Sorry, no progress yet