NVIDIA-Omniverse/PhysX #192: Controller Manager flags not working as expected

cpyburn, 2023-09-08:
When having multiple controllers with...
```c++
shape->setFlag(PxShapeFlag::eSIMULATION_SHAPE, true);
shape->setFlag(PxShapeFlag::eSCENE_QUERY_SHAPE, true);
```
under one controller manager and then setting one controller to...
```c++
shape->setFlag(PxShapeFlag::eSIMULATION_SHAPE, false);
shape->setFlag(PxShapeFlag::eSCENE_QUERY_SHAPE, false);
```
all the other controllers still collide with it.  This all works fine when every controller has its own controller manager, but isn't the point of a "manager" is to control multiple objects?  Not sure if this is a bug or expected behavior but seems like a bug.

### Library and Version
PhysX v5.1.0

### Operating System
Windows 10

PierreTerdiman, 2023-09-19:
The shape flags that you mention are for PxShape objects. It is unrelated to controllers, which are PxController objects.

See PxControllerFilterCallback for defining how PxController interact with each other, and more generally speaking PxControllerFilters.

cpyburn, 2023-09-20:
Well then why does it work for individual controller managers but not one controller manager with multiple controllers? And PxController has a PxShape, so this is very confusing... and not consistent behavior if nothing else.

PierreTerdiman, 2023-09-20:
I think we didn't expect people to use multiple controller managers.

What happens is that during a move() call, only the controllers registered in the controller manager are considered for collisions:

[](https://github.com/NVIDIA-Omniverse/PhysX/blob/main/physx/source/physxcharacterkinematic/src/CctCharacterController.cpp#L2400
)

However, "PxController has a PxShape" indeed, and I assume these underlying kinematic actors/shapes all belong to the same PxScene. So while we do not consider the PxControllers for collisions, the system still sees their PxActor/PxShape. And it fails to filter them away because the data used to do that filtering is per-controller-manager as well:

[](https://github.com/NVIDIA-Omniverse/PhysX/blob/main/physx/source/physxcharacterkinematic/src/CctCharacterControllerCallbacks.cpp#L998)

[](https://github.com/NVIDIA-Omniverse/PhysX/blob/main/physx/source/physxcharacterkinematic/src/CctCharacterController.cpp#L2535)

So what happens is:

- with a single controller manager the PxShape flags are ignored and the CCT-vs-CCT filtering is handled by the filtering callback I previously mentioned.

- with "individual controller managers" that filtering callback does nothing, because each manager only sees a single controller. Instead we end up colliding against the PxScene actors from other controllers, that we fail to filter out. At this point, indeed, the PxShape flags that you asked about end up controlling the filtering. But this is unexpected, not by design.

So, why are you using multiple controller managers? Is there a use-case for this?

cpyburn, 2023-09-20:
When the Shape filters didn't work, I tried using multiple controllers, and it worked and thought that was by design.

With your explanation it all makes sense now.  I will revert my code back to one controller manager and use the CCT-vs-CCT filtering.  Maybe you could make the PxShape for PxController be private?  Could possibly help with anyone else that thinks the shape would work.  Just a thought.

closed 2023-09-20 by cpyburn