NVIDIA-Omniverse/PhysX #195: PxCreateCudaContextManager issue

ENGY-ORG, 2023-09-18:
Perhaps I am linking wrong but whenever I try to enable GPU Rigid Dynamics it gives me the following messages:

```
"demoGame.exe" (Win32): "H:\NGE\ngeTools\demoGame\_debug\PhysXDevice64.dll" loaded. 
RDOC 001728: [19:24:57]      nvapi_hooks.cpp( 519) - Log     - Initialised nvapi, version NVidia Complete Version 1.10
RDOC 001728: [19:24:57]      nvapi_hooks.cpp( 519) - Log     - Initialised nvapi, version NVidia Complete Version 1.10
"demoGame.exe" (Win32): "H:\NGE\ngeTools\demoGame\_debug\PhysXDevice64.dll" unloaded.
C:\b\w\2a49d215dfbd0a40\source\cudamanager\src\PhysXDeviceSettings.cpp (191) : warning : PhysXDevice64.dll is not available, should be in the same directory as PhysXGpu.dll

C:\b\w\2a49d215dfbd0a40\source\cudamanager\src\CudaContextManager.cpp (653) : info : No PhysX capable GPU suggested.

C:\b\w\2a49d215dfbd0a40\source\cudamanager\src\CudaContextManager.cpp (654) : info : If you have a PhysX capable GPU, verify that PhysX is not set to CPU in the NVIDIA Control Panel.
```
at:
`gCudaContextManager = PxCreateCudaContextManager(*mFoundation, cudaContextManagerDesc, PxGetProfilerCallback());`

It also does not conncet to the Visual Debugger.
But everything else seems to work fine.

The same code works well when directly compiled with the application. But when I have it in an external static lib it gives me these messages.

Thank you for the help.

preist-nvidia, 2023-10-27:
Did you check if the device DLL is in the same directory as the gpu dll? See the warning?

preist-nvidia, 2023-12-11:
Closing for now, please reopen if we should pick this up again.