NVIDIA-Omniverse/PhysX #200: [Joints] Broken joints, when removed/re-added to the scene come back unbroken

NBG-Juozas, 2023-10-06:
I've noticed that ScConstraintSim.cpp
Sc::ConstraintSim::ConstraintSim(ConstraintCore& core, RigidCore* r0, RigidCore* r1, Scene& scene) :

sets the constraint flags to eBREAKABLE and eCHECK_MAX_FORCE_EXCEEDED, but eBROKEN does not get set, even if the "Core" class has that constraint flag raised.

If the eBROKEN flag is activated here, based on that core instance, then a lot of places trip asserts, since they expect the joint not to be broken.

### Library and Version
PhysX v4.1.2
But from simple code inspection it seems that this issue should be present in Px5 as well

### Operating System

Windows 10

### Steps to Trigger Behavior

 1. Create a Scene, PxRigidActor and PxJoint that attaches to that actor
   1.1 Add a breaking force to the PxJoint so that it's breakable
 3. Add PxRigidActor with joint to the scene
 4. Add enough force to break the joint
 5. Remove actor from scene
 6. Add actor to scene again, the joint will add forces, make the rigid actor get constrained and possibly break the joint again, even though the PxConstraintFlags still say that the joint is broken

### Expected Behavior

Removing and readding a broken joint should retain the "Broken" status

### Actual Behavior

Removing and readding a broken joint seems to restore the unbroken status in the low level code to some degree, though higher level flags still say that that joint is "broken"

msauter-nvidia, 2023-10-11:
Hi,
A ticket has been created to define more clearly what the expected behavior is in this scenario and ensure it works as defined. For now, I would recommend to not re-use a broken joint but rather create it from scratch again.

NBG-Juozas, 2023-10-11:
Thanks for the response. 

Does the PhysX public repo (v4 or v5) have any available test suite to possibly verify any workarounds/hotfixes done locally?

preist-nvidia, 2023-10-11:
Unfortunately not, our testing suite is internal only.

preist-nvidia, 2023-12-11:
Internal tracking: PX-4475