NVIDIA-Omniverse/PhysX #201: Flaw in error handling of HeightField creation

Tisten, 2023-10-10:
I got an error about "PxHeightField::release: double deletion detected" and so I got started on a memory goose hunt in vain. The issue was bad data sent into MeshFactory::createHeightField() and the reported error fooled me.

### Library and Version
PhysX v5.3.0

### Operating System
Windows 11

### Steps to Trigger Behavior
 1. Call `MeshFactory::createHeightField(PxInputStream& stream)` with bad data, e.g wrong header.

### Code Snippet to Reproduce Behavior
```
auto physx = PxCreatePhysics(PX_PHYSICS_VERSION, foundation, PxTolerancesScale());
PxDefaultMemoryInputData input((PxU8*)("invalid_input"), 13);
auto heightfield = physx->createHeightField(input);
```

### Expected Behavior
Nullptr returned from createHeightField(). Possibly with an error about an invalid header emitted.

### Actual Behavior
Getting `"PxHeightField::release: double deletion detected!"` coming from https://github.com/NVIDIA-Omniverse/PhysX/blob/a2af52eb6a2532bd2bc583ef8ead9c81c9222af1/physx/source/geomutils/src/GuMeshFactory.h#L191

There is no "double delete", the problem is that the heightfield object can't be removed from the factory since it had not been added yet.

### Suggested fix
In these lines:
https://github.com/NVIDIA-Omniverse/PhysX/blob/a2af52eb6a2532bd2bc583ef8ead9c81c9222af1/physx/source/geomutils/src/GuMeshFactory.cpp#L1012-L1018
Move `addHeightField(np);` up so it is done before `if(!np->load(stream))`. That way the `MeshFactory::onRefCountZero()` will be able to call `mf->remove(*object)` and the "double delete" error wont be issued.

JC-Jumyung, 2023-10-12:
Thank you for your feedback. You are correct that the error message is misleading. In such cases, we will directly delete the object instead of repositioning addHeightField, eliminating the double deletion error.

Tisten, 2023-10-13:
Deleting it directly sounds fine.

And I guess you should change the code in the same way for `MeshFactory::createConvexMesh()` since the pattern looks the same.

preist-nvidia, 2023-10-27:
Fix will be in next release.

closed 2023-10-27 by preist-nvidia