NVIDIA-Omniverse/PhysX #206: The object does not continue its trajectory after deactivation of constraint

ThinkingThreeD, 2023-10-27:
I'm trying to integrate PhysX into a Maya plugin

Although I know that the engines are not all the same, I have different behavior between Bullet and PhysX on a relatively simple exercise.

I have an animated rigid body cube, and another active rigid body constrained to it (PX6Joint with Drive)

frame 0, the constraint is activated (Limited)  and frame 4, I deactivate the constraint  (Free) .
In bullet the object continues its trajectory but in physX it falls horizontally (under gravity)
Is there a setting to activate to have the same behavior as Bullet?

Thanks

Bullet
[image]

PhysX
[image]

ThinkingThreeD, 2023-10-29:
The same exercise works well in PVD so I think it's an integration problem in Maya
I will look at it
[image]

preist-nvidia, 2023-10-30:
Thank you for the update! Please close the issue when you confirm that it's an integration and not an SDK issue.

ThinkingThreeD, 2023-11-02:
Yes thank you @preist-nvidia

ThinkingThreeD, 2023-11-03:
Hi,
Just to be sure that this is the right method, because otherwise I think we really have a problem.
Currently if I want my object to continue its trajectory after the constraint is deactivated I do this:

[image]

I deactivate the kinetimatic at the same time as the deactivation of the joint
[image]

If I disable the constraint only, the object falls straight down
Is this the ideal method?

```c++
#define PX_PHYSX_STATIC_LIB
#include <iostream>

#include <ctype.h>
#include <cmath>
#include <algorithm>
#include <vector>

#include "PxPhysicsAPI.h"
#include "foundation/PxPreprocessor.h"

#define PVD_HOST "127.0.0.1"
#define PX_RELEASE(x)   if(x)   { x->release(); x = NULL;   }

using namespace physx;

PxDefaultAllocator      gAllocator;
PxDefaultErrorCallback  gErrorCallback;

PxFoundation* gFoundation = NULL;
PxPhysics* gPhysics = NULL;

PxDefaultCpuDispatcher* gDispatcher = NULL;
PxScene* gScene = NULL;

PxMaterial* gMaterial = NULL;

PxPvd* gPvd = NULL;

PxD6Joint* joint = NULL;
PxRigidDynamic* parent = NULL;

enum InterpolationType
{
    LINEAR,
    STEP
};

template <typename T>
struct KeyFrame
{
    float time;
    T value;
    InterpolationType interpType;
};

template <typename T>
class AnimCurve
{
public:
    AnimCurve() {}

    template <typename U>
    void AddKeyframe(float time, U value, InterpolationType interpType = LINEAR)
    {
        KeyFrame<U> key;
        key.time = time;
[... 208 lines cut ...]

}

int main(int, const char* const*) {
    static const int frameCount = 500;

    rigid_translateX.AddKey(0, 0.0f);
    rigid_translateX.AddKey(30, 20.0f);

    joint_enabled.AddKey(0, true, InterpolationType::STEP);
    joint_enabled.AddKey(5, false, InterpolationType::STEP);

    initPhysics();
    createBoucingBall();

    for (int i = 0; i < frameCount; i++)
    {
        stepPhysics(i);
    }

    cleanupPhysics();

    return 0;
}
```

ThinkingThreeD, 2023-11-03:
The following screenshot of the code shows how in Bullet I do to obtain the same animation
[image]

[image]

[image]

ThinkingThreeD, 2023-11-04:
The same method doesn't work in Omniverse so maybe I'm using it like Bullet when there's another way that I don't know of to maintain the velocity
[image]

ThinkingThreeD, 2023-11-08:
Hi,
Can you tell me if the re-opening of the issue is visible, because I have no response from you.
Thanks

preist-nvidia, 2023-11-08:
Yes, I see this, but have not had the chance to respond yet.

I'm still a bit confused by the setup. Do you have a kinematic that you switch to dynamic and want the velocity to persist, or do you have a dynamic that is jointed to a kinematic via D6 and when you disable the joint the dynamic does not keep the velocity?

preist-nvidia, 2023-11-08:
Ah ok now I get it - yes I am not sure that works well. Can you try this please:
- Animate a kinematic rigid body
- Attach the dynamic to the kinematic with a D6
- Disable the constraint when you want to let go

ThinkingThreeD, 2023-11-08:
Thank you!!
It's this technique naturally that I did when I closed this issue, but I wanted to make sure that the Bullet method is not compatible (or rather definitely not the same) with PhysX

preist-nvidia, 2023-11-08:
Got it - yes kinematic-D6-rigid dynamic is the way to go!

Closing again then.

closed 2023-11-08 by preist-nvidia