NVIDIA-Omniverse/PhysX #207: BLAST: Island removal doesn't work as expected

TwentyPast4, 2023-11-04:
### Library and Version
[Blast SDK 5.0.2](https://github.com/NVIDIA-Omniverse/PhysX/releases/tag/105.0-blast-5.0.2)

### Operating System
Windows 11

### Steps to Trigger Behavior

There are two ways to observe the issue
 1. Enable automatic island removal (NvBlastExtAuthoringFractureTool::setRemoveIslands)
 2. Load a mesh that will most likely lead to mesh islands when fractured, something like a ladder
 3. Slice it down the middle twice to remove the ladder's side rails
 4. The rungs of the ladder will be scaled and positioned incorrectly (that is, the geometry in Nv::Blast::AuthoringResult)

Or:
 1. Do not enable automatic island removal
 2. Load a mesh that will most likely lead to mesh islands when fractured, something like a ladder
 3. Slice it down the middle twice to remove the ladder's side rails
 4. Manually call NvBlastExtAuthoringFractureTool::islandDetectionAndRemoving using the middle chunk's ID (the one that now has mesh islands)
 5. The rungs of the ladder will be scaled and positioned incorrectly (that is, the geometry in Nv::Blast::AuthoringResult)

### Code Snippet to Reproduce Behavior
It is way to complicated to post here, but boils down to

```
fractureTool->setRemoveIslands(true);
// ... generate the mesh to be used by blast ...
int32_t chunkId = fractureTool->setChunkMesh(mesh, -1);
Nv::Blast::SlicingConfiguration slicingConf;
slicingConf.x_slices = 3; // assuming ladder had rungs going along X axis
// assuming a rng exists with name anyRandomGenerator
fractureTool->slicing(chunkId, true, slicingConf, &anyRandomGenerator);

// assuming a ConvexMeshBuilder exists with name cmb
Nv::Blast::BlastBondGenerator* bondGen = NvBlastExtAuthoringCreateBondGenerator(&CollisionBuilder);
Nv::Blast::ConvexDecompositionParams params;
params.maximumNumberOfHulls = 1; // to skip the wait time of this as it is irrelevant
Nv::Blast::AuthoringResult* resultingGeometry = NvBlastExtAuthoringProcessFracture(*fractureTool, *bondGen, *cmb, params);
// results in resultingGeometry->geometry are incorrect!
```

### Expected Behavior
The scaling and offsets of the resulting chunks should be correct.

### Actual Behavior
The scaling and offsets of chunks are incorrect, and thus are rendered and physically represented incorrectly (chunks are either too big or too small, and in wrong locations)

LaBouette, 2023-11-05:
I noticed this too when I tried to remove islands after first fracture (for things like windows). Thanks for the fix, hope it will land in this repo soon.

nvidia-bgaldrikian, 2023-12-20:
I added a comment to [https://github.com/NVIDIA-Omniverse/PhysX/pull/208](https://github.com/NVIDIA-Omniverse/PhysX/pull/208) which gives my proposed solution, and explains why I looked for an alternative solution.  Please have a look and if that works for you, I'll move forward with it.

Thank you,
Bryan

TwentyPast4, 2023-12-25:
The proposed solution by [bgaldrikian](https://github.com/nvidia-bgaldrikian) works in my test cases as well.

closed 2024-01-26 by preist-nvidia