NVIDIA-Omniverse/PhysX #209: PhysX sdk 2.8.4.5 ?

wanderingies, 2023-11-07:
Please， where can I download PhysX sdk 2.8.4.5 msi ?

WWebber, 2023-11-18:
This seem to be a 3dsmax version only. I know only V2.8.4.4. For what you need it ?

wanderingies, 2023-11-18:
> This seem to be a 3dsmax version only. I know only V2.8.4.4. For what you need it ?

DdsTrans ，This source code needs to use this library, Trying 2.8.4.4 and 2.8.4.6 are both wrong

WWebber, 2023-11-18:
I can't imagine that the changes between these versions are that big that they can't be fixed in the source code, if you have it.
There's also an old thread about it: [https://forums.developer.nvidia.com/t/physx-sdk-2-8-5-download/28718](url)

wanderingies, 2023-11-19:
> I can't imagine that the changes between these versions are that big that they can't be fixed in the source code, if you have it. There's also an old thread about it: [https://forums.developer.nvidia.com/t/physx-sdk-2-8-5-download/28718](url)

thanks, this link is no longer valid