NVIDIA-Omniverse/PhysX #210: Trigger shapes not detecting TOUCH_LOST when going through PlaneShape

lpares12, 2023-11-10:
### Library and Version

PhysX v5.1.3

### Operating System

Ubuntu 20.04

### Steps to Trigger Behavior

 1. Create a RigidDynamic body with one trigger shape.
 2. Create a RigidStatic body with one normal/contact Plane shape and set it below the RigidDynamic body
 3. Set the custom filter shader to the code below
 4. Run the simulation and check for pairs->status flags on SimulationEventCallback::onTrigger() callback

### Code Snippet to Reproduce Behavior

```
PxFilterFlags px_filter_shader(PxFilterObjectAttributes attributes0,
        PxFilterData filter_data0,
        PxFilterObjectAttributes attributes1,
        PxFilterData filter_data1,
        PxPairFlags& pair_flags,
        const void* constant_block,
        uint32_t constant_block_size) {
    if(filter_data0.word0 & filter_data1.word1 && filter_data1.word0 & filter_data0.word1) {
        if(PxFilterObjectIsTrigger(attributes0) || PxFilterObjectIsTrigger(attributes1)) {
            pair_flags = PxPairFlag::eTRIGGER_DEFAULT;
        }else{
            pair_flags = PxPairFlag::eCONTACT_DEFAULT;
        }
        pair_flags |= PxPairFlag::eNOTIFY_TOUCH_FOUND;
        pair_flags |= PxPairFlag::eNOTIFY_TOUCH_LOST;
    } else {
        return PxFilterFlag::eSUPPRESS;
    }

    return PxFilterFlag::eDEFAULT;
}
```

### Expected Behavior

I would expect the onTrigger callback to be called twice. Once when the RigidDynamic body contacts the plane for the first time. And one when it exits. The first one should have TOUCH_FOUND flag set, and the second one TOUCH_LOST.

### Actual Behavior

Callback is called only once with TOUCH_FOUND flag, and never with TOUCH_LOST

preist-nvidia, 2023-11-16:
Internal tracking PX-4582

PierreTerdiman, 2024-01-08:
This may be expected:

https://nvidia-omniverse.github.io/PhysX/physx/5.3.1/docs/Geometry.html#planes
"Planes divide space into “above” and “below” them. Everything “below” the plane will collide with it."

So if you let a primitive drop under gravity "below" the plane, the collision is still detected and there shouldn't be any "lost contact" event. You would only get a lost event when the primitive goes back up above the plane.

lpares12, 2024-01-08:
Makes sense, thanks!

PierreTerdiman, 2024-01-09:
I wrote a test and confirmed that the code works as expected, as far as I can tell. Closing this for now, please re-open if you think there is a real issue.

closed 2024-01-09 by PierreTerdiman