NVIDIA-Omniverse/PhysX #225: CCD report a contact point far from the shape

pkusilence, 2023-12-06:
Library and Version : PhysX v3.4.0 (I checked that CCD and gjkraycast is not changed in PhysX5)
Operating System: iOS 17.1.2

We opened the Macro CCD_ANGULAR_IMPULSE, and sometimes the contact point from CCD can be far from the rigid body, and the rigid body will bounce with a huge linear velocity ( w x r is too large since r is large). After debuging for days, I found that for function GJKCPairDoSimplex, when the input size is 3 and with Q0 = Q1, for example : 

 0 Q A B =-2569.475830078125,  -1834.175537109375,  -205.80770874023438, 
-2351.701416015625,  -2024.861572265625,  -192.2308349609375,
 217.77435302734375,-190.68597412109375,13.576877593994141 

1 Q A B =-2569.475830078125,  -1834.175537109375,  -205.80770874023438, 
-2351.701416015625,  -2024.861572265625,  -192.2308349609375, 
217.77435302734375,  -190.68597412109375,  13.576877593994141 

 2 Q A B =6690.65576171875,  4742.841796875,  335.57254028320312, 
6507.9609375,  4560.80859375,  349.1494140625,
 -182.69482421875,  -182.03305053710938,  13.576885223388672 

the inner function closestPtPointTriangle checked the triangle(simplex set) but always return Q0 and Q1, the later closestPtPointSegment returns Q0.

So is this a bug, or is there any mechanism to ensure the input Q points are not same? Besides, I tried but can never reproduce this case (i.e. get same Qs in CCD)  on Windows/Android.

preist-nvidia, 2023-12-07:
Thanks for the report, @pkusilence  - we'll have a look. Internal tracking: PX-4666.

pkusilence, 2023-12-11:
It turns out that the original tiangle has a huge edge length ratio, i.e La = Lb >>>Lc. Due to the float point differences between platforms, the support function found different vertices if the search direction is nearly perpendicular to La /Lb。
[image]

nvtw, 2024-01-09:
Unfortunately huge edge length differences (sliver triangles) are notoriously problematic especially with single precision. If you can avoid such triangles by e. g. using a different tessellation, that would be great. Debugging is difficult for us unless you could provide the full code to reproduce the described issue.

pkusilence, 2024-01-24:
> Unfortunately huge edge length differences (sliver triangles) are notoriously problematic especially with single precision. If you can avoid such triangles by e. g. using a different tessellation, [...]

I fixed it by checking wheather Q0 == Q1 .
[image]

Indeed we need to avoid such annoying triangles, thanks for your advice!

closed 2024-01-24 by pkusilence