NVIDIA-Omniverse/PhysX #226: Crash when there is too much collision happening

Elercia, 2023-12-06:
### Library and Version

PhysX 5.3.0 (last commit on trunk)

### Operating System
Windows 10 

### Steps to Trigger Behavior

I know this repro is a little bit hacky and kind of limit testing but it trigger a crash caused by an out-of-memory / out-of-block inside the ``PxcNpContext::mNpCacheStreamPair``.

The simpliest steps I found to reproduce are:
1. Create 5000 rigidbody at the exact same position
2. Attach for each a shape built from a custom geometry (like PxCustomGeometryExt::CylinderCallbacks)
2. Run a simulation step

### Code Snippet to Reproduce Behavior

Starting from SnippetHelloWorld ``initPhysics`` function. 
```
for( int i = 0; i < 5000; i++ )
{
	PxCustomGeometryExt::CylinderCallbacks* cylinder = new PxCustomGeometryExt::CylinderCallbacks( 10.f, 10.f, 0, 1.f );

	PxRigidDynamic* actor = gPhysics->createRigidDynamic( PxTransform( 0.f, 0.f, 0.f ) );
	PxRigidActorExt::createExclusiveShape( *actor, PxCustomGeometry( *cylinder ), *gMaterial );
	gScene->addActor( *actor );
}

gScene->simulate( 1.0f / 60.0f );
gScene->fetchResults( true );
```

This is triggering a crash inside ``discreteNarrowPhase`` (PxcNpBatch.cpp(445)) because a pointer is not tested. 

### Expected Behavior
As the documentation says [here](https://nvidia-omniverse.github.io/PhysX/physx/5.1.0/docs/Simulation.html#k-data-blocks), the simulation should drop contacts & joints constraints in this case. 
The behavior is known and defined so the pointer should be tested, isn't it? I can see other places like inside `` physx::writeCompressedContact`` where pointers are in fact tested.

**I have a question related to this issue:**
How can we facilitate the debugging of such behaviors. Do you have a mechanism of limit report that we can use in this kind of situation ?
Something like a callback logging every contacts made so far so we can analyse what is happenning when a allocator is running out of memory / when we have too much contacts.

### Actual Behavior
Pointer is not tested after expected failed allocation.

preist-nvidia, 2023-12-07:
Thank you for the report and repro @Elercia ! We'll have a look. Internal tracking: PX-4665

adenzler-nvidia, 2024-01-11:
Thanks again for the report - we have fix for this coming in the next release. You will get a warning when you hit the limit.

In the meantime, I suggest increasing the max number of contact blocks, or not instantiating as many rigid bodies in the same location. As for better debugging: our policy is that overflowing any preallocated structure should output at least a debug warning. If this is not happening in some case, please file another issue and we will have a look.

closed 2024-01-11 by adenzler-nvidia