NVIDIA-Omniverse/PhysX #229: Generating project files doesn't succeed due to missing dependencies

DawidNoras, 2023-12-11:
### Library and Version

PhysX v5.3.0

### Operating System

Windows 10

### Steps to Trigger Behavior

 1. Clone https://github.com/NVIDIA-Omniverse/PhysX.git
 2. enter physx
 3. run generate_projects.bat
 4. see provided message

### Expected Behavior

Project files should be properly generated

### Actual Behavior

Projects files are not generated due to the issue. Generation process is aborted

See output after running generate_projects.bat

> Preset parameter required, available presets:
> (0) vc15win64 <--- VC15 Win64 PhysX general settings
> (1) vc16win64 <--- VC16 Win64 PhysX general settings
> (2) vc17win64 <--- VC17 Win64 PhysX general settings
> Enter preset number: 2
> Running generate_projects.bat vc17win64
> Running packman in preparation for cmake ...
> Processing project file 'N:\Test\PhysX\src\physx\dependencies.xml'
> Package 'clang-physxmetadata' at version '4.0.0.32489833_1' is missing from local storage.
> Downloading from https://d4i3qtqj3r0z5.cloudfront.net/clang-physxmetadata%404.0.0.32489833_1
> .       (speed 896 KiB/s)
> Total of 0.00 seconds
> packman(ERROR): Tar archive extension must be one of: [.tar .tar.bz2 .tar.bzip2 .tar.gz .tar.gzip .tar.xz]
> '\VsWhere.exe' is not recognized as an internal or external command,
> operable program or batch file.
> '\VsWhere.exe' is not recognized as an internal or external command,
> operable program or batch file.
> '\VsWhere.exe' is not recognized as an internal or external command,
> operable program or batch file.
> Using preset xml: buildtools/presets/public/vc17win64.xml
> Target platform: win64 using compiler: vc17
> Traceback (most recent call last):
>   File "N:\Test\PhysX\src\physx\buildtools\cmake_generate_projects.py", line 321, in <module>
>     main()
>   File "N:\Test\PhysX\src\physx\buildtools\cmake_generate_projects.py", line 317, in main
>     presetProvided(presetName)
>   File "N:\Test\PhysX\src\physx\buildtools\cmake_generate_projects.py", line 248, in presetProvided
>     print('PM_PATHS: ' + os.environ['PM_PATHS'])
>   File "N:\packman-repo\python\3.7.9-windows-x86_64\lib\os.py", line 681, in __getitem__
>     raise KeyError(key) from None
> KeyError: 'PM_PATHS'
> Hit ENTER to continue...

SumantBagri, 2023-12-11:
Can confirm that this is happening for the linux preset as well
```
./generate_projects.sh linux
5.966 Total of 0.00 seconds
#35 5.967 packman(ERROR): Tar archive extension must be one of: [.tar .tar.bz2 .tar.bzip2 .tar.gz .tar.gzip .tar.xz]
#35 5.988 Traceback (most recent call last):
#35 5.988   File "./buildtools/cmake_generate_projects.py", line 321, in <module>
#35 5.988     main()
#35 5.988   File "./buildtools/cmake_generate_projects.py", line 312, in main
#35 5.988     result = subprocess.run(['bash', './generate_projects.sh', presetName], cwd=os.environ['PHYSX_ROOT_DIR'], check=True, universal_newlines=True)
#35 5.988   File "/root/packman-repo/python/3.7.9-linux-x86_64/lib/python3.7/subprocess.py", line 512, in run
#35 5.988     output=stdout, stderr=stderr)
#35 5.988 subprocess.CalledProcessError: Command '['bash', './generate_projects.sh', 'linux']' returned non-zero exit status 1.
```

spham-amzn, 2023-12-11:
Looks like an issue with cloudfront, the link that is attempting to download [clang-physxmetadata](https://d2cerybs4newgl.cloudfront.net/clang-physxmetadata%404.0.0.32489833_1) is returning a `403` error:

```
403 ERROR
The request could not be satisfied.
Request blocked. We can't connect to the server for this app or website at this time. There might be too much traffic or a configuration error. Try again later, or contact the app or website owner.
If you provide content to customers through CloudFront, you can find steps to troubleshoot and help prevent this error by reviewing the CloudFront documentation.

Generated by cloudfront (CloudFront)
Request ID: gHidmCt8AY10laN9ra_K9ILzGNgFbyUpgp2AROp6FNfxOWWMbP-4nQ==

```

vegata, 2023-12-12:
https://d4i3qtqj3r0z5.cloudfront.net/XXXXXXXXXXXX   cannot connect

preist-nvidia, 2023-12-12:
Yes this is a problem from our side, working on a fix. Will update asap.

preist-nvidia, 2023-12-12:
The issue is with Packman. Please do the following:

1. Navigate to ``physx/buildtools/packman``
2. Run ``./packman update -y``
3. ``generate_projects.bat|sh`` will now succeed in downloading the dependencies.

Apologies for the inconvenience. We will push 5.3.1 to main today/tomorrow which will fix this.

DawidNoras, 2023-12-12:
updating packman worked. 

Thank you

preist-nvidia, 2023-12-12:
5.3.1 landed and will fix this. For older commits/release versions, please continue using the manual packman update described here. Will also add an announcement for this.

closed 2023-12-12 by preist-nvidia