NVIDIA-Omniverse/PhysX #238: Position Based Dynamics: Set Particle Position and Velocity

ErlerPhilipp, 2024-01-27:
### Library and Version

Physx 5.3.1

### Operating System

Windows 10

### Issue

The [documentation for PBD](https://nvidia-omniverse.github.io/PhysX/physx/5.3.1/docs/ParticleSystem.html#position-based-dynamics-pbd-particle-systems) lacks some information:
How can I set the position and velocity of a particle? A snippet for a particle emitter would be great.

Context:
I'm trying to implement a water gun. It should select unused or hidden fluid particles and teleport them to the emitter.

I tried modifying the array returned by PxParticleBuffer::getNbActiveParticles() but it raises an error. I guess that this array is not meant to be written into, maybe related to CUDA memory allocation and syncing.

preist-nvidia, 2024-01-29:
Hi @ErlerPhilipp,

Check the documentation again, just a bit lower where the particle buffer and its update flags are discussed. This is how you can update positions and velocities from a buffer, and how you can implement your watergun by either creating and adding new particle buffers to the system, or better, creating all particles initially but park them somewhere invisible and then teleport them to the emitter location.

Hope this helps,
Philipp

ErlerPhilipp, 2024-01-29:
Hi @preist-nvidia,

yes, thanks. I don't know how I missed this. Maybe, I thought that was only for initialization.

Related question: Is there a way to directly interact with the particle buffer from a custom CUDA kernel, without the detour through host memory? Would I need to run my kernel in the PhysX CUDA context or could this safely work between 2 contexts?

preist-nvidia, 2024-01-29:
Hi @ErlerPhilipp - yes of course you can interact with the buffers from your own CUDA code.

We think it'll be easier to work with a single CUDA context (either use the PhysX one or create the PhysX context with your own); but it should also be possible to work with two contexts.

Philipp

ErlerPhilipp, 2024-01-29:
All right. Thanks!

closed 2024-01-29 by ErlerPhilipp