NVIDIA-Omniverse/PhysX #239: TGS instability with drives

kermado, 2024-02-02:
Hello,

I've been running into a few stability issues when using the TGS solver. I was able to reproduce the simplest example that exhibits the oscillation that I've been seeing.

See video here:
https://github.com/NVIDIA-Omniverse/PhysX/assets/120369/3f133a9e-2f99-4915-8d38-f08948f704ae

The scene consists of two bodies. The first is jointed to the scene and the second is jointed to the first. Both joints have drives, with target positions and velocities set to zero.

Everything works perfectly when using the PGS solver. I noticed quite a few stability issues when the center of mass is quite far away from the joint position, even when drives were not used. However, in this example, the center of mass is not so far away from the joint anchor and adding drives causes the second body to oscillate.

Is there anything that can be done to improve this without the usual tricks (repositioning the joints, changing the center of masses or fudging the inertia tensors)?

### Library and Version
PhysX v5.3.1

### Operating System
Windows 10

### Steps to Trigger Behavior

 1. Modify SnippetHelloWorld.cpp with the code here: https://gist.github.com/kermado/3fe59c0ec1dcc43988f288bec4657d2a
 2. In SnippetHelloWorldRender.cpp, modify camera position: `sCamera = new Snippets::Camera(PxVec3(2.0f, 2.0f, 2.0f), PxVec3(-0.6f,-0.2f,-0.7f));`

### Code Snippet to Reproduce Behavior
Snippet provided here: https://gist.github.com/kermado/3fe59c0ec1dcc43988f288bec4657d2a

### Description
Bodies don't oscillate.

### Actual Behavior
Body oscillates.

kermado, 2024-02-02:
It seems to behave a lot better if preprocessing is disabled.

msauter-nvidia, 2024-02-06:
Hi kermado,

Thanks for the report and the nice repro. On our latest internal build, I do not see the instability. Could you maybe quickly double check whether the posted repro is exactly the one you meant to attach? If yes, then it looks like the problem might have been fixed. I do recall us fixing a bug in TGS related to preprocessing, so it could potentially be related. For now, I can only suggest to wait for the next release and then see if this solves the instabilities beyond this repro too.

kermado, 2024-02-06:
Thanks. Yeah, that’s the full repro. I have no other changes. I’ll check again once the next release is published.