NVIDIA-Omniverse/PhysX #242: PxRigidBodyExt::setMassAndUpdateInertia of kinematic rigid actor with triangle mesh  causes: Assertion failed: mG.isFinite()

nathalie-raffray, 2024-02-16:
### Library and Version
PhysX v5.3.1

### Operating System
Windows 11

### Steps to Trigger Behavior

 1. Create a dynamic actor
 2. Set it to kinematic
 3. Create a triangular mesh shape and assign it to this actor
 4. Call PxRigidBodyExt::setMassAndUpdateInertia with a mass of 1 and center of mass (0,0,0)

### Code Snippet to Reproduce Behavior

```
const auto pDynamicBody = pPhysics->createRigidDynamic(globalTransform);
// Set kinematic
pDynamicBody->setRigidBodyFlag(PxRigidBodyFlag::eKINEMATIC, true);

// Assign triangular mesh shape to pDynamicBody
const auto pTriangleMesh = pPhysics->createTriangleMesh(input);
const auto pMaterial = pPhysics->createMaterial(0.5f, 0.5f, 0.0f);
PxRigidActorExt::createExclusiveShape(*pRigidActor, PxTriangleMeshGeometry(pTriangleMesh, 1.0f), *pMaterial);

// Set mass and update inertia
const auto mass = 1.0f;
const auto centerOfMass = glm::vec3(0.0f, 0.0f, 0.0f);
PxRigidBodyExt::setMassAndUpdateInertia(*pDynamicBody, mass, &centerOfMass);
```

### Expected Behavior

The behavior for kinematic triangle mesh actor should be the same as the for other kinematic actors with other shapes (there is no assertion failing, i.e. I don't have this issue with convex meshes, boxes, spheres, etc).

### Actual Behavior

`PX_ASSERT` fails and my program running in Debug stops with the following error:
ExtInertiaTensor.h(282) : Assertion failed: mG.isFinite()

[image]
It seems like the issue is caused by the mass of the triangle mesh being 0 here (lines ~200) in ExtRigidBodyExt.cpp. 
I debugged and the mass, inertia and centerOfMass are default values (i.e. 0 for mass and inertia).
This mass is later used in ExtInertiaTensor.h, and mG is calculated by dividing by that mass (which equals 0) which leads to mG not being finite.

preist-nvidia, 2024-02-16:
Thank you for the detailed report, I'll have a look. Internal tracking: PX-4820

preist-nvidia, 2024-04-23:
Hi @nathalie-raffray - sorry for the late reply.

Two points:
- For dynamic/kinematic rigid bodies, you need to add an SDF to the triangle mesh. You should get a message.
- Do you get the assert on any input mesh? In theory, I think we should be able to compute the mass of a mesh.

I tried quickly to update our SnippetSDF to print the mass of the tri-mesh geom after running update mass, and this works. Even if I set the bowl dynamic to kinematic.

preist-nvidia, 2024-07-25:
Closing due to inactivity.

closed 2024-07-25 by preist-nvidia