NVIDIA-Omniverse/PhysX #247: SDF cause CUDA error , even black screen (driver crash?)

axilesoft, 2024-02-27:
### Library and Version

PhysX v5.3.1

### Operating System

Windows 11

### Steps to Trigger Behavior

 1.  create a SDF rigidbody , and lots of simple rigidbodies ( non-KINEMATIC ), some connected with joints
 2.  do collision with SDF 
 3.  change some rigid body to eKINEMATIC

### Expected Behavior

normal collision

### Actual Behavior

when change some rigid body to eKINEMATIC, shen getResults, will report ERROR:
code: eINTERNAL_ERROR
message: PhysX Internal CUDA error. Simulation can not continue! Error code 700!
file: C:\b\w\eb2f45c4acc808a0\physx\source\gpusolver\src\PxgSolverCore.cpp
line: 527

preist-nvidia, 2024-02-27:
Hi @axilesoft - thank you for the report, but this is hard to diagnose without a specific repro. Could you please provide the repro steps you describe in a snippet that compiles?

axilesoft, 2024-02-28:
I'm not sure repro steps, the eKINEMATIC change have higher rate to repro, but seems other thing may affact too, like gpu ecoding.
I want to know what is PxgSolverCore.cpp line 527 about. Then I can try some ways to avoid it.

preist-nvidia, 2024-03-04:
The error code you get is not directly helpful, as it indicates an illegal memory access in the SDK. So it's something we need to track down and fix. If you can provide a repro in a snippet that compiles, we can do that.

preist-nvidia, 2024-04-23:
Closing for now, as no repro was provided. Please reopen with a repro if still an issue.

closed 2024-04-23 by preist-nvidia