NVIDIA-Omniverse/PhysX #252: Increasing Damping ingnores ForceLimit of PxD6 Joint

HerrDerSchoepfung, 2024-03-11:
I have created a PxD6joint and tested different values. Therefore, it is unclear why, with a target speed of 1m/s and a maximum force limit of 1, an object with a mass of 1kg can move upwards when the damping parameter is edited (above 8).
[image]
Furthermore, it is not possible to move the object upwards if the maximum force is unlimited and a target speed of 1m/s is set when damping is set to 1.

[image]
[image]

### Library and Version

PhysX v5.3.1

### Operating System

Windows 10

### Code Snippet to Reproduce Behavior

The Update part gets called every step.
```
PxJoint* ConstraintHandler::AddSliderConstraint(V3SSliderConstraint^ constraint, PxRigidActor* actor0, PxRigidActor* actor1)
{
	pxScene->lockWrite();
	PxTransform frame0 = Helper::V3SMatrixToPxTransform(constraint->FrameA);
	PxTransform frame1 = Helper::V3SMatrixToPxTransform(constraint->FrameB);
	auto pxjoint = PxD6JointCreate(*pxPhysics, actor0, frame0, actor1, frame1);

	if (pxjoint == nullptr)
	{
		pxScene->unlockWrite();
		throw gcnew PhysicEngineException("Can not create the point to point constraint!");
	}

	if (constraint->AreLimitsEnabled) 
	{
		pxjoint->setMotion(PxD6Axis::eX, PxD6Motion::eLIMITED);
		pxjoint->setLinearLimit(PxD6Axis::eX, PxJointLinearLimitPair(constraint->LowerLimit, constraint->UpperLimit, PxSpring(0, 0)));	
	}
	else
	{
		pxjoint->setMotion(PxD6Axis::eX, PxD6Motion::eFREE);	
	}

	auto drive = PxD6JointDrive(PxZero, constraint->Damping, constraint->ForceLimit, true);
	pxjoint->setDrive(PxD6Drive::eX, drive);

	pxjoint->setDriveVelocity(PxVec3(constraint->TargetSpeed, 0, 0), PxVec3(PxZero));

#ifdef _DEBUG
	pxjoint->setConstraintFlag(PxConstraintFlag::eVISUALIZATION, true);
#endif
	pxScene->unlockWrite();
	return pxjoint;
}

void ConstraintHandler::UpdateSliderConstraint(V3SConstraint^ constraint)
{	
	pxScene->lockWrite();
	auto sliderConstraint = safe_cast<V3SSliderConstraint^>(constraint);
	auto pxjoint = constraint2physx[constraint]->GetPxJoint()->is<PxD6Joint>();

	if (pxjoint == nullptr)
	{
		pxScene->unlockWrite();
		throw gcnew PhysicEngineException("Can not create the point to point constraint!");
	}

	auto frame0 = Helper::V3SMatrixToPxTransform(sliderConstraint->FrameA);
	auto frame1 = Helper::V3SMatrixToPxTransform(sliderConstraint->FrameB);

	if (sliderConstraint->AreLimitsEnabled)
	{
		pxjoint->setMotion(PxD6Axis::eX, PxD6Motion::eLIMITED);
		pxjoint->setLinearLimit(PxD6Axis::eX, PxJointLinearLimitPair(sliderConstraint->LowerLimit, sliderConstraint->UpperLimit, PxSpring(0, 0)));
	}
	else
	{
		pxjoint->setMotion(PxD6Axis::eX, PxD6Motion::eFREE);
	}

	PxD6JointDrive drive(PxZero, sliderConstraint->Damping, sliderConstraint->ForceLimit, true);
	pxjoint->setDrive(PxD6Drive::eX, drive);
	pxjoint->setDriveVelocity(PxVec3(sliderConstraint->TargetSpeed, 0, 0), PxVec3(PxZero));

	pxjoint->setLocalPose(physx::PxJointActorIndex::eACTOR0, frame0);
	pxjoint->setLocalPose(physx::PxJointActorIndex::eACTOR1, frame1);

	pxScene->unlockWrite();
}
```

### Expected Behavior

Mass: 1kg
Damping: 1 
TargetSpeed: 1 
ForceLimit: infinity 

Object should move upwards

Mass: 1kg
Damping: 9 
TargetSpeed: 1 
ForceLimit: 1 

Object should not move upwards

### Actual Behavior

Mass: 1kg
Damping: 1 
TargetSpeed: 1 
ForceLimit: infinity 

Object does not move

Mass: 1kg
Damping: 9 
TargetSpeed: 1 
ForceLimit: 1 

Object moves upwards

Kenny-Vilella, 2024-03-15:
Hello,

Thank you for the report !
I will have a look. Internal tracking: PX-4879.

HerrDerSchoepfung, 2024-04-02:
Are there any findings regarding this topic or do you need more information? # @Kenny-Vilella

Kenny-Vilella, 2024-04-02:
My apologies for the delay.
I have a few other issues that requires my attention.
As soon as I could start looking into it, I would let you know.

Kenny-Vilella, 2024-04-09:
Hello,
I had time to take a quick look.
I believe that the snippet of code you provided cannot be run on its own.
Without a small repro it is difficult to provide a final answer.
Nevertheless, please find below some observations that may be useful:
- The force limit is converted internally as an impulse limit (force x timestep), so with a force limit of 1.0 you may still expect the box to move depending on the selected time-step. If you multiply the force limit by the time step, the box will probably not move independently of the conditions.
- The behavior of the drive will highly depends on the stiffness/damping used. In particular, if you use a low stiffness, the drive is likely to not reach the target velocity/position. Increasing the stiffness should normally lead to a better tracking.

Kenny-Vilella, 2024-04-09:
I will investigate a bit more the first point on the conversion between force limit and impulse limit and let you know.

Kenny-Vilella, 2024-04-09:
After investigation, by default the `driveForceLimit` provided is actually an impulse limit.
To switch to an actual force limit, it is necessary to turn on the `PxConstraintFlag::eDRIVE_LIMITS_ARE_FORCES` flag.
This can be done by calling:
`pxjoint->setConstraintFlag(PxConstraintFlag::eDRIVE_LIMITS_ARE_FORCES, true);`

HerrDerSchoepfung, 2024-04-16:
Thanks for your help. I will test it and get back to you afterwards.

HerrDerSchoepfung, 2024-04-16:
I played around with the stiffness a bit, but this only resulted in my objects bouncing back and forth.
It is also noticeable that with unlimited force, a damping of 1 and a target speed of 1 m/s, the target speed should basically be reached almost immediately, but with a simulation time of 10 seconds, this is only reached from approx. 5.

My goal is to be able to move objects by specifying a speed and a maximum force.  For example, objects standing in the way would stop the moving object if the maximum force is not sufficient or, if the maximum force is sufficient, the objects would also be displaced.

Something like this simplified formula would do:
```
var mass = MovingObject3D.Mass.Value;
var timesteps = 0.01;
var maxForce
var forceNeeded = mass * ((Speed - constraint.Speed) / timesteps);

if (forceNeeded > 0)
{
    PhysicManipulator.ApplyForce(MovingObject3D, globalTranslationAxis * maxForce);
}
```

Setting the ```PxConstraintFlag::eDRIVE_LIMITS_ARE_FORCES``` also does not change my situation.
Am I missing something critical or is there a better solution to achieve this?

Kenny-Vilella, 2024-05-15:
My apologies for the delay.

Setting the `PxConstraintFlag::eDRIVE_LIMITS_ARE_FORCES` would make sure that the force limit is actually applied, and I believe that it is working properly.

> I played around with the stiffness a bit, but this only resulted in my objects bouncing back and forth.
> It is also noticeable that with unlimited force, a damping of 1 and a target speed of 1 m/s, [...]

It all depends on the parameter of your drive. Setting a high value for the damping would make the drive much more efficient at reaching the target speed.
You should select the drive stiffness and damping depending on the desired behavior. If you want a drive that follows precisely the target position/velocity then high stiffness and damping would be better.

I played a bit and could reproduce the behavior that you want.
Put a box on a plane and actuated another box with a drive on a PxD6Joint, depending on the force limit given to the drive the actuated box could move or be stopped by the other box.

closed 2024-08-27 by HerrDerSchoepfung