NVIDIA-Omniverse/PhysX #259: Trigger emulation and character controller

karjonas, 2024-03-21:
### Library and Version

PhysX v5.3.1

### Operating System

Linux

### Steps to Trigger Behavior

It is suggested in the documentation that you can emulate trigger bodies using shader filters (https://nvidia-omniverse.github.io/PhysX/physx/5.3.1/docs/AdvancedCollisionDetection.html#triggers). However, these do not seem to work with character controllers. Is this expect or a bug? I have modified the SnippetTriggers snippet to use a character controller instead. I have attached a patch file for the SnippetTriggers you can apply to try it out.

[fallingchar.txt](https://github.com/NVIDIA-Omniverse/PhysX/files/14705035/fallingchar.txt)

PierreTerdiman, 2024-03-25:
Character controllers use kinematic actors and scene-queries, so there are some changes needed in the snippet to be compatible with that system.

a) tell PhysX to keep interactions between kinematic and static actors when creating the scene:

`	sceneDesc.staticKineFilteringMode = PxPairFilteringMode::eKEEP;
`

Without this, the overlaps between static & kinematic actors are discarded by the broadphase, and the filter shader emulating the trigger behavior is never called (the character controller silently goes through the trigger).

b) disable scene-queries on trigger shapes in the "filter callback" case:

```
	else if(impl==FILTER_CALLBACK)
	{
		PxShapeFlags shapeFlags = PxShapeFlag::eVISUALIZATION | PxShapeFlag::eSIMULATION_SHAPE;
		// We will have access to shape pointers in the filter callback so we just mark triggers in an arbitrary way here,
		// for example using the shape's userData.
		shape = gPhysics->createShape(geom, *gMaterial, isExclusive, shapeFlags);
		shape->userData = shape;	// Arbitrary rule: it's a trigger if non null
	}
```

Without this, the character controller's shape registers an impact against the trigger, and the chatacter stops against the trigger. This is only needed in the "filter callback" codepath:

- the character controller systems already discards regular triggers so this is not needed for that codepath
(https://github.com/NVIDIA-Omniverse/PhysX/blob/main/physx/source/physxcharacterkinematic/src/CctCharacterControllerCallbacks.cpp#L1002).

- the "filter shader" codepath in the snippet already omits PxShapeFlag::eSCENE_QUERY_SHAPE when creating the trigger shape, so the character controllers do not see the shape and do not collide against them.

----

After these two changes, the snippet should work with character controllers.

karjonas, 2024-03-26:
Thanks for the information

closed 2024-03-26 by PierreTerdiman