NVIDIA-Omniverse/PhysX #265: Unable to compile with Clang

thempc, 2024-04-01:
### PhysX v5.3.1
### Windows 11

### Steps to Trigger Behavior
 1. Build PhysX
 2. Include the PxPhysicsAPI.h in an external project
 3. Try to build external project using the platform toolset "LLVM (clang-cl)" in Visual Studio Community 2022 - Version 17.8.5

### Code Snippet to Reproduce Behavior
```
#include <iostream>
#include <PhysX/PxPhysicsAPI.h>

int main() {
    std::cout << "Hello, world!" << std::endl;
    return 0;
}
```

### Expected Behavior
It's expected the code compiles without issues.

### Actual Behavior
The code will fail to compile and gives the following errors:

```
1>In file included from ..\..\..\Source\Boot\Main.cpp:46:
1>In file included from ..\..\..\includes\PhysX/PxPhysicsAPI.h:99:
1>In file included from ..\..\..\includes\physx/foundation/PxVecMath.h:1327:
1>In file included from ..\..\..\includes\physx/foundation/PxInlineAoS.h:36:
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3088,8): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3088,32): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3088,50): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3089,8): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3089,32): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3089,50): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3090,8): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3090,32): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3090,50): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3091,8): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3091,32): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3091,50): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3092,8): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3092,32): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3092,50): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3093,8): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3093,32): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3093,50): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>..\..\..\includes\physx/foundation/windows/PxWindowsInlineAoS.h(3094,8): error : member reference base type 'VecU16V' (aka '__m128') is not a structure or union
1>fatal error: too many errors emitted, stopping now [-ferror-limit=]
```

preist-nvidia, 2024-04-02:
Hi @thempc - we only test and support MSVC on Windows is that not an option for you?

On Linux we support Clang, [see the platform readme](https://github.com/NVIDIA-Omniverse/PhysX/blob/main/physx/documentation/platformreadme/linux/README_LINUX.md), but you may have to update some build configuration settings/defines to make Clang work on Windows.

preist-nvidia, 2024-04-02:
Yes, always best to let us know! And we do need to state more clearly what compilers we support on Win.

Please don't worry about the PR for now, I'd ping you if it becomes relevant. Thank you very much for offering!

Closing with this.

closed 2024-04-02 by preist-nvidia

zopsicle, 2024-08-07:
It may still be worth noting that Microsoft (vendor of MSVC) recommends against accessing the fields of `__m128` directly:

https://learn.microsoft.com/en-us/cpp/cpp/m128?view=msvc-170

> You should not access the `__m128` fields directly.

There's only a handful places where PhysX does this, so it may be worth fixing regardless.