NVIDIA-Omniverse/PhysX #275: I can not link physx 5.3.0 library from vcpkg into my project

ChivenZhang, 2024-04-23:
### Library and Version

PhysX v5.1.0

### Operating System

Windows 10, visual studio 2022, v142 toolset

### Steps to Trigger Behavior

 1. install physx:x64-windows using vcpkg
 2. link 15 libs into project
 3. compile failed as lnk2019 error

### Code Snippet to Reproduce Behavior

[image]

[image]

[image]

[image]

### Expected Behavior

compile pass

### Actual Behavior

compile failed

preist-nvidia, 2024-04-25:
Hi @ChivenZhang - if you follow the instructions here, does it work? https://omni-101.github.io/omniverse/2023/07/07/omniverse-physx-vcpkg-port/

ChivenZhang, 2024-04-25:
It do works, Thanks. But I wonder why I can not use physx in ordinary way, just install the library and link it.

preist-nvidia, 2024-06-13:
Hi @ChivenZhang - sorry we cannot help with linker/build config beyond pointing you to the unofficial vcpkg port docs.

closed 2024-06-13 by preist-nvidia