NVIDIA-Omniverse/PhysX #276: Static contact forces are not correct on inclined plane

cga-cmu, 2024-04-28:
### Library and Version
PhysX-105.1-blast-5.0.4

### Operating System
Ubuntu 22.04

### Steps to Trigger Behavior

 1. Put block on slope and collect contact forces

### Code Snippet to Reproduce Behavior

```
/*******************************************************************
Test physx contact forces vs. acceleration, Does F=ma?
*******************************************************************/

#include <stdio.h>
#include <stdlib.h>
#include <ctype.h>
#include <iostream>
#include <vector>
#include "PxPhysicsAPI.h"
#include "snippetcommon/SnippetPrint.h"
#include "snippetcommon/SnippetPVD.h"
#include "snippetutils/SnippetUtils.h"
#include "snippetrender/SnippetRender.h"
#include "snippetrender/SnippetCamera.h"

/******************************************************************/

using namespace std;
using namespace physx;

/******************************************************************/
// Defines

#define N_XYZ 3
#define XX 0
#define YY 1
#define ZZ 2

// physx has multiple contacts
#define MAX_N_CONTACTS 4

// Typically we use a time step of 1/60 of a second (60Hz) and 10 iterations.
// 8/3 is also good. This provides a high quality simulation in most games
// float timeStep = 1.0f / 60.0f;
#define TIME_STEP (1.0f/60.0f)

/******************************************************************/
/******************************************************************/
// Globals

// Physx stuff
static PxDefaultAllocator gAllocator;
static PxDefaultErrorCallback gErrorCallback;
static PxFoundation* gFoundation = NULL;
static PxPhysics* gPhysics = NULL;
static PxDefaultCpuDispatcher* gDispatcher = NULL;
static PxScene* gScene = NULL;
static PxMaterial* gMaterial = NULL;
static PxPvd* gPvd = NULL;

std::vector<PxVec3> gContactPositions;
std::vector<PxVec3> gContactNormals;
std::vector<PxVec3> gContactImpulses;
std::vector<PxReal> gContactSeparations;

PxRigidDynamic *box_body;
[... 432 lines cut ...]
  */

  return 0;
}

/******************************************************************/
```

### Expected Behavior
Fx = 0
Fy = 9.81

### Actual Behavior
Static forces are wrong.
For a block not moving on a 30 degree slope
with default PxMaterialFlags
Fx = 4.2479 (should be zero)
Fy = 7.3575 (should be g, which is set to 9.81)
The overall force magnitude is too big (sqrt(Fx*Fx + Fy*Fy) > 9.81), 
so it can't be that the forces are reported in some rotated coordinate system.
Various combinations of strong friction, improved
patch friction, and compliant contacts did not
substantially change these numbers.

Contact forces are also wrong for a wheel rolling down a fixed slope or rolling on a curved slope (a large sphere).

cga-cmu, 2024-04-28:
Box2D gets all these cases (static box on slope, wheel rolling down a fixed slope, and wheel rolling on a curved slope (a large sphere) right.

preist-nvidia, 2024-04-29:
Thank you for the detailed report and repro, @cga-cmu ! We will investigate. Internal Tracking PX-5048.

farbod-farshidian, 2024-05-02:
There is an issue with the current release of Physx. The Contact Reporter only reports the normal force and does not report the friction force. Therefore, the force that it measures is  

`F = m g cos(\theta) \hat{n}`

where `\hat{n}` is the surface normal. Therefore, the reported force will be

`F = m g [cos(\theta) sin(\theta), cos^2(\theta)]`. 

For m=1.0Kg, this will be F = [4.24, 7.35]. Therefore, the reported value is correct.

I have heard that this issue will be fixed in the next release, and the API will report the friction forces.

msauter-nvidia, 2024-05-14:
It's as farbod-farshidian is pointing out. The impulses reported in contact reports are only the impulses applied along the contact normals. The next version of PhysX will provide an API to also extract the friction impulses. If then summed up, it will converge to the expected y=9.81 force (plus/minus 0.01 or so) in your example (once it settles down after the initial impact of the falling box).

preist-nvidia, 2024-05-14:
@cga-cmu - we expect to release the update with the friction reporting by the end of the month.

closed 2024-05-14 by preist-nvidia