NVIDIA-Omniverse/PhysX #279: Bad task scheduling due to lack of priority

Zeblote, 2024-05-12:
Here's an example. CM and ABP tasks are scheduled at roughly the same time. CM task is usually quite fine grained, while ABP has few very long ones. But the engine has no way to know this! So it often schedules them terribly.

[image]

We can fix this by letting the engine know the ABP task has higher priority. Here's the result of passing `ETaskPriority::High` to new `UE::Tasks` API in Unreal for ABP tasks only.

[image]

I only implemented the API necessary to get the info to Unreal, not in the default task scheduler, so it's probably not worthy of a PR as is.
https://github.com/brickadia/PhysX/commit/cde092e8a1f0a297ea9b6b74686d589033d8004e

Zeblote, 2024-05-12:
On that note, I'm wondering if it is possible to further split the PABP tasks. It seems like the dynamics (full) one is constantly taking much longer than the others, but I don't really understand the internal code that well.

PierreTerdiman, 2024-05-15:
Thank you for the report. This is interesting, and I think we could try that in our own benchmarks. AFAIK the scheduler does not support task priorities so we would need to implement that first.

There is currently no way to further split the PABP tasks. This is possible in theory though, and there is / was some experimental code doing just that a while ago. But it has never been productized (IIRC it introduced more task-related overhead, to the point that only really large scenes showed benefits). It would be interesting to revisit this, but this is probably not a high-priority task.

PierreTerdiman, 2024-12-23:
Hello,

Support for high-priority tasks has now been added to the default CPU dispatcher, and it is now used by the broadphase tasks. I can confirm that I see small performance improvements after the change. So, thank you for that. The change should appear in the next release.

Additionally, while digging into that code again for this work, I found an issue that could also affect performance: we constantly resize an internal array in ABP_PairManager::resizeForNewPairs(). This is unexpected, and frankly a bug (my mistake). This should only resize the array when it's full (like the original single threaded code) This can become quite expensive for large scenes with a lot of pairs. I will submit a bugfix for that one after the Christmas holidays. Meanwhile you can change the code like this, it's a very simple change:

```
void ABP_PairManager::resizeForNewPairs(PxU32 nbDelayedPairs)
{
	PxU32 currentNbPairs = mNbActivePairs;

	const PxU32 newNbPairs = currentNbPairs + nbDelayedPairs;

	const PxU32 newHashSize = PxNextPowerOfTwo(newNbPairs + 1);
	if(newHashSize == mHashSize)
		return;

	// Get more entries
	mHashSize = newHashSize;
	mMask = newHashSize - 1;

...
```

closed 2025-01-03 by PierreTerdiman