NVIDIA-Omniverse/PhysX #281: How to read data from Flow

ManosAgelidis, 2024-05-13:
Hi, I have been trying to extract data from Flow by using the snippet below (and similar snippets writing the output of the simulation to the filesystem) provided in the NvFlowEditor app
[image]

Unfortunately the file that is written is in binary form. I tried validating it with nanoVBD's nano_validate tool and convert it to OpenVBD format using nanoVBD's nano_convert tool but it fails with
[image]

I suppose the reason for the failure is that the file is in binary format. Can you please clarify whether there is a way to programmatically extract data from Flow to use in other applications e.g., in Blender? And if so, what would be the correct pipeline from Flow to the external app? Thanks a lot!

areidmeyer, 2024-06-18:
This turned out to be a bug in NanoVDB itself unfortunately. A future update will allow raw NanoVDBs written out like this to work as is.

In the meantime, it is still possible to take the buffer coming out of Flow, construct a NanoVDB handle, and save that to file. That will also allow you to do things like enable zip compression if desired.

ManosAgelidis, 2024-06-19:
HI Andrew, 

Thanks for the response. I actually managed to read the raw data from the filesystem with these NanoVDB lines:
[image]

The problem is that of course writing to disk and then reading the raw NanoVDB file is super costly. I would like to understand how to directly pass the raw buffer to NanoVDB, is there perhaps a code snippet in Flow that I missed? I tried various methods, like writing the buffer to a isstringstream, or to raw memory buffers inheriting from std::streambuf that are then passed to std::istream instances but NanoVDB was never happy with any of the efforts that I made when reading the streams instead of a file. Any suggestions? Thanks a lot

areidmeyer, 2024-06-25:
nanovdb::HostBuffer test_buffer;
	test_buffer.createFull(smokeNanoVdbReadbackSize, smokeNanoVdbReadback);
	nanovdb::GridHandle<nanovdb::HostBuffer> test_handle{ test_buffer };
	
The buffer that Flow outputs can be used to construct a nanovdb GridHandle directly, then you can use NanoVDB from there on the data, copy free.

ManosAgelidis, 2024-06-25:
Amazing, will try it, thanks a lot!

closed 2024-06-25 by ManosAgelidis