NVIDIA-Omniverse/PhysX #286: D6 Joint doesn't report drive force

kermado, 2024-06-08:
It looks like the drive force isn't reported for some joints. In particular, the D6 joint doesn't report the drive force/torque. Force reporting is only enabled for locked axes. I can't see any way to enable drive force reporting without modifying the source code.

Would it be possible to add a flag to `PxD6JointDriveFlag` or similar? This could then be used to set the `Px1DConstraintFlag::eOUTPUT_FORCE` in `ConstraintHelper::addDrive()`.

vreutskyy, 2024-06-13:
Hi @kermado 
Thanks for the reporting this issue. We'll check it.

preist-nvidia, 2024-06-27:
@kermado - can you use an articulation instead? We do have https://nvidia-omniverse.github.io/PhysX/physx/5.4.0/docs/Articulations.html#link-incoming-joint-force

And if you project this onto the DOF, you'll get the drive force (plus friction/limits if they are active).

kermado, 2024-06-30:
@preist-nvidia I can in most cases, but sometimes we have cycles that need to be broken and we still need to measure forces/torques for those joints. At the moment, I have made the changes described, which allows me to get these values out.

preist-nvidia, 2024-07-01:
Understood - would you mind creating a PR for the changes and we can look at the code and consider integrating it?

msauter-nvidia, 2024-08-07:
Update: support for adding the drive forces is planned to go into release 5.5.0-106.2. Look for PxD6JointDriveFlag::eOUTPUT_FORCE

closed 2024-11-07 by preist-nvidia