NVIDIA-Omniverse/PhysX #287: Enabling eMODIFY_CONTACTS causes jitter on wall collision.

kujukuju, 2024-06-20:
### Library and Version

Physx 5.1.0

### Operating System

Windows 11

### Steps to Trigger Behavior

 1. Set up a scene with a wall
 2. Create a dynamic rigid body
 3. Enable the scene flag eMODIFY_CONTACTS (you don't actually have to modify anything)
 4. Apply a CHANGE_VELOCITY force to steadily move your dynamic rigid body into the wall
 5. See the jittering effect

### Expected Behavior

I expect the dynamic rigid body to not be jittery. When you disable eMODIFY_CONTACTS it's smooth.

### Actual Behavior

Jittery when colliding with walls.

---

I'm trying to manually manipulate some contacts in my game, so I've set up a contact modify callback and enabled the eMODIFY_CONTACTS scene flag. I'm using the rust version of physx which is 5.1.

For reference, I am able to modify contacts and things work as expected.

However, when I enable modify contacts all collisions in my scene where I push up against a wall begin to jitter. This happens even when I do completely nothing in my PxContactModifyCallback.

I couldn't find another reported issue on this github about this. From what I can tell, it's because modifying contacts causes the physics engine to generate new patches.

It seems like this is an existing issue, because I found this user on a forum who reported my exact issue.
https://gamedev.net/forums/topic/715265-physx-rigidbody-jitters-if-pushed-into-corners-depenetration-issue/

He ended up just disabling contact modification and stopped there. So unfortunately not helpful for me.

I actually went into the source code and commented out the area where it seems to force a patch update when contacts are modified and this did nothing. I wanted to go through all the source code step by step and find exactly where the bug was being caused but I actually couldn't find anywhere eMODIFY_CONTACTS was used to perform additional steps that seemed to affect the jitter bug at all.

However, I can confirm as shown in my video, as with the gamedev.dev guy, the bug only happens when the contact modification flag is enabled. Nothing else matters at all.

Here is a video showing the bug. I have eMODIFY_CONTACTS enabled at the start (with no actual contact modification happening at all), and once it's disabled the bug goes away.

https://github.com/NVIDIA-Omniverse/PhysX/assets/5142257/2b594934-bc01-4b32-8a20-da560845b287

Thanks!

kujukuju, 2024-06-21:
This issue seems to be caused by the contacts generated during the modify contact pass having a restitution that's not based on the material values of the shapes colliding. If you go through and set every contact to it's expected restitution (and friction?) value in every modify contact callback the issue resolves itself.

I think this is a real bug. Having non-expected contact properties in the modify callback forces those values back into the real simulation if you don't correct them yourself.

Even with this fix the simulation isn't quite as smooth as with modify contacts disabled. I think this is probably as good as I can get it now, but if there are any further recommendations for how to stabilize my scene that'd be appreciated.

vreutskyy, 2024-06-24:
Hi @kujukuju 
Thanks for the detailed report. I've created an internal issue. We'll check it and fix in one of the next releases.